bk1285/rpi_wordclock #93: How to address web socket correctly

JonahPi, 2018-07-07:
Hi all, I'd like to use the web-socket to control the wordclock from Node-Red but I don't know how to address it correctly. I see that my connection request reaches the Raspberry but the connection is immediately terminated without any error message. From the client side I get the message: "ws:undefined : Error: not opened". As URL I use "ws://192.168.0.58:8081". Beside this I also don't know exactly how the payload of the JSON file should look like, I've seen that there is one parameter "API" which should be 2 but I have no idea what this means. Has anybody a sample Websocket request which e.g. activates the pluging "sunrise" ?
Cheers
Jo

bk1285, 2018-07-08:
Hi @JonahPi 
this is currently a bit experimental. Best practice would be to use the version on feature/refactoring branch
The according interfaces are listed here: https://github.com/bk1285/rpi_wordclock/blob/feature/refactoring/wordclock_interfaces/web_interface.py 
* "/set/something"
* "/get/something"
Example, how it's used is here https://github.com/bk1285/rpi_wordclock/blob/feature/refactoring/wordclock_interfaces/static/js/form.js

Best,
Bernd

JonahPi, 2018-07-08:
Hi Bernd
Unfortunately that doesn't help me much since I am a beginner in terms of python programming.
Not sure if you know Node-Red but it makes it usually pretty easy to connect to Websocket, you have Web-socket client node (which I configured to connect to ws://192.168.0.58:8081) and then you use an inject node for the payload (I selected to inject a JSON formated payload )
However the file wordclock_socket.py exits in the part below, it seems self.request.recv doesn't receive any data

>     def recv_msg(self):
        # Read message length and unpack it into an integer
        raw_msglen = self.__recvall(4)
        if not raw_msglen:
            raise
        msglen = struct.unpack('>I', raw_msglen)[0]
        # Read the message data
        return self.__recvall(msglen)

    def __recvall(self, n):
        # Helper function to recv n bytes or return None if EOF is hit
        data = ''
        while len(data) < n:
            packet = self.request.recv(n - len(data))
            if not packet:
                return None
            data += packet
        return data

.

Cheers
Jo