bk1285/rpi_wordclock #94: Buttons

crob11, 2018-07-16:
Hallo Bernd,

ich habe die Wordclock nach Anleitung gebaut inkl. den 3 Buttons. Auch der Autostart wurde eingepflegt. Den Code habe ich nicht verändert. 

Zum Problem: 
Die Buttons hatten beim ersten Start sogar mal funktioniert (konnte verschiedene Farben wechseln etc.). Nach Trennen der Stromzufuhr und wiederverbinden funktionieren die Buttons nicht mehr (keine Funktion) und die Wordclock arbeitet stets auf einer Farbe. Die Verschaltung habe ich mehrfach überprüft und keinen Fehler gefunden (habe auch nix verändert).

Wo liegt mein Fehler? Für jegliche Ideen/Hilfe vielen Dank im Voraus!

schweizschweiz, 2018-07-26:
Auch mit einem Multimeter gemessen?

crob11, 2018-07-28:
Falls du die Spannung über den Button meinst,... ja. 
An den Buttons liegen jeweils 3,3V an (kommt ja ausm GPIO und müsste passen)

schweizschweiz, 2018-08-01:
Und die Spannung fällt auch auf 0v wenn du einen gedrückt hältst? Also es klingt ja eher nach Hardware als Software Fehler oder nicht?

bk1285, 2018-09-16:
Hi @crob11 

klingt für mich ebenfalls nach HW-Fehler... Ggf. einfach mal den develop-branch ohne HW-buttons (via webinterface) testen.

Best,
Bernd

FrankX0, 2018-09-20:
> Nach Trennen der Stromzufuhr und wiederverbinden funktionieren die Buttons nicht mehr.
This certainly sounds like a hardware issue.
This might be introduced by a noisy power-supply. Some of the (cheap) Chinese power supplies for LED applications produce a lot of noise, which in my case e.g. led to flickering of the LED's during operation.

Can you share your wiring diagram and maybe pictures?

bk1285, 2021-07-25:
Please reopen, if still unresolved.

closed 2021-07-25 by bk1285