bk1285/rpi_wordclock #102: Problems using raspberry zero w - distortet output

atec111, 2018-11-04:
Hi all,

I've come to this project from microcontroller.net version of a wordclock with usage of STM32F103C8T6 and ESP. I've built my watch after there tutorial and it worked for some time, but always after a while it just hangs (not that smart for a smartlike clock ;-)). I assume the problem was the ESP and my Wifi. But never found out.

I was very happy hearing of this project, because i like raspberries a lot more than arduinos or ESPs. But unfortunately it did not work out of the box for me :-(

Symptom:
Application starts without errors. But Time as Words is lighted in different colors the same time, and not at the right place or amount (tried both wirings in the config). Other Plugins are also not working correctly. The best result comes with "rainbow" - it shows a correct horizontal Rainbow, but with decent flickering with wrong colors from time to time (more in the upper half, than in bottom).
But the good thing - it seems all LEDs can blink =)

What I did:
Almost everything what needs to be done on an brandnew from the scratch zero W. I've also done everything from Troubleshooting (using level shifter, same ground) and from closed threads (autostart in cron and no hello message).

My Wiring (as in Mikrocontroller.net):
[image]

A Good shot from the Rainbow:
[image]

And one from Time as Words (german):
[image]

I've just an 5V/4A Power Fuse, could that be the Problem?

Regards

FrankX0, 2018-11-04:
These symptoms suggest a noise problem, normally associated with not using a level shifter and/or a noise power-supply. As you already are using a level shifter this leaves the power supply.
For my clock I initially used a cheap Chinese power supply, which caused similar problems and even destroyed the first LED of the clock due to excessive noise.
Can you try to power your clock with a (5V) power-bank: if this works correctly you need to search for a new (maybe lower power rating) power supply. I use a 3A type. A lower power rating will also be less noisy.

Hope this helps.

bk1285, 2018-11-04:
Hi @atec111 

I suspect at least two issues
#1: A hardware issue as mentioned by @FrankX0 
#2: The need of a correct wiring. I updated the develop branch according to your wiring posted aboved: https://github.com/bk1285/rpi_wordclock/commit/a394b1d15079d601f75d70eb59ea3f57907ce7af 
Ensure to set the wiring_layout to micro_net_wiring within your config-file.

Best,
Bernd

atec111, 2018-11-06:
Hi all,

Thanks Bernd, your Wiring-Layout helped a lot. The correct words/letters are lit now.
But additionally I assume that my power supply is not the perfect one. I was testing the clock when it was attached to a "power-cube" (multiple power plugs), and it seems, that some other device caused the worst of the distortions - when I attached it back to it's target destination (a single power-plug), the flickering/ghosting was mostly gone.

Now also the animations like sunrise work, despite some little misslighting of some letters from time to time.

I think it is necessary to change the power supply as the next step. Are there any recommendations? (Clock is mounted approximately 1,8m above a power-plug)

Regards,
Andre

atec111, 2018-11-12:
Update:
I've soldered an Micro-USB-Female Port to my Circuit and tryed a Power Bank (Anker, 5V/2A). Result: It was worse than my prior 5V/4A Power Supply. More flickering in Rainbow, more wrong lit lights in "the time".
I've even tried a strong Micro-USB Power-Supply (Anker, with 5V/2.4A per Port). Result: Even more worse than the Power Bank. In Rainbow there was almost no correct picture and it was flickering so slow, that it was kind of wandering of single pixels - was looking kind of space invaders :-)

I've put my old Power-Supply back to the Clock. I've read in the web, that some guys powering a WS2812 could obeserve that the flickering would minimize, when using full brightness. I also tested this - with very good results.
With brightness be less than 100 = Not usable. With brightness at max (255) very very few flickers in rainbow, and they getting even fewer with the time the clock is powered.

So far so good. Clock is usable, but full brightness is not the best solution. **Anyone any Idea what could be done to solve the same result in the same way?** I've read on the web, that using a Capacitor (bigger=better) on Data-Pin could/should help... Others say a 1K Resistor should. But there is never a really plausible electrical statement why that should help...

Regards,
Andre

FrankX0, 2018-11-12:
Hi Andre,

As the power supply seems to be sufficient, I would guess it is due to the wiring.
Can you share in detail how you have wired your clock, starting from the power source?

When you set the clock to a stable (non changing) display, do you the also see the flickering? Or do yo only see flickering when the contents changes (like the rainbow)?
Can you disconnect the data line after a LED patterns is set? And then check for flickering?
@bk1285: do you only send new data when required (so in case of the clock only every minute)?

For your information: I am running my application also with a zero W without any flickering.

HeerNMeester, 2018-11-16:
I had similar problems after a power outage.
I did not use a level shifter before but got the same results after installing one. In the end I replaced my RPi Zero V1.3 for a RPi Zero W. That solved the problem. My conclusion: the Pi died during the power outage or at least something on the GPIO.

FrankX0, 2018-11-16:
My guess would be that the dataline from the Pi to the level shifter and/or from the level shifter to the first LED is being disturbed. Either by external noise or by distortion due to return currents.
If not done already, can you run a 0V connection close to the dataline from the Pi I/O connector to the 0V of the level shifter and from the level shifter to the 0V of the first LED?

bk1285, 2018-11-18:
>  @bk1285: do you only send new data when required (so in case of the clock only every minute)?
Yes, only once per minute.

FrankX0, 2018-12-09:
@atec111 did you succeed resolving your issue?

phenze, 2018-12-13:
Hey guys,

i had this issue several times in the last two years since i built my clock.
The Clock is working and sometimes colors are very strange and flickering around.
Then after a while everything is normal again. Then again flickering.

I do not use a level shifter. As a power supply i use 5V/10A.

In my case it was everytime a broken led in the stripe or wrong connections between the stripes.
I think i have a very bad stripe.
I don't know but a broken LED could lead to a wrong data signal ? Couldn't it ? 
I'm not a hardware specialist.
That also explains why the rainbow plugin is working best. There the data is sent very very often so that a wrong signal produced by a broken LED will not have that much effect.

But in Bernds wordclock software the signal is only sent every minute. 

My solution was the following :

Did you try the strandtest example ? 
In my case i have modified it a bit that only the color wipe(turning ON each led individually ) is done. 
When running the example i can see exactly which LED has the error.

After that i cut the strip and replaced the broken LED. Then everything is running well again.

A modified strandtest example i have uploaded here:
https://github.com/bk1285/rpi_wordclock/issues/106

Tetrikus, 2018-12-13:
Same for me, when I've built my own Wordclock. I do use capacitors and LevelShifter, but whenever I have these issues, it's one of the following:

- Hardware defect of one LED
- Loose contact on the GPIO-Header
- Loose soldering contact of the signal

atec111, 2018-12-13:
I assume the same for my clock. But I think it is more a loose contact somewhere on data-wire.
But unfortunately I hadn't enough time for detailed analysis until now. I hope xmas holidays will give me some ;-)
I'll keep you updated.

bk1285, 2021-07-24:
Reopen, if still unresolved.

closed 2021-07-24 by bk1285