bk1285/rpi_wordclock #105: Wrong LEDs light up

EstherJohanna, 2018-12-11:
Hallo Bernd, vielen Dank für die ausführliche Anleitung! Die Wordclock ist unser erstes Raspi-Projekt und wir haben die LEDs tatsächlich zum Leuchten gebracht. Leider zeigen sie nicht die richtigen Wörter an. Nun fragen wir uns, an welcher Stelle wir etwas falsch gemacht haben. Wir haben dein Wiring benutzt, aber keine Buttons angebracht. Das Foto zeigt unseren Aufbau. Viele Grüße, Laura und Esther 

[image]

bk1285, 2018-12-12:
Hallo Laura, hallo Esther,

auf dem Bild sind die 110 LEDs für die Buchstaben zu sehen, die 4 Minuten LEDs fehlen da noch: 2 direkt am Raspberry, 2 am Ende des LED-Strips.

Wenn's danach noch Probleme gibt, einfach nochmal melden.

Viel Erfolg!
Bernd

bk1285, 2018-12-12:
... siehe auch: https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_layouts/german_11x10/wiring_front_bernds_wiring.png

bk1285, 2018-12-30:
Bei weiteren Fragen issue bitte wieder öffnen.

closed 2018-12-30 by bk1285