bk1285/rpi_wordclock #106: Crazy Lights gone wild

MaxZimmermann99, 2018-12-13:
Hallo und erstmal Danke für das tolle Projekt!

Ich sitz hier mit einem Raspberry Zero und hab alles verlötet. Ganz egal was ich aber versuche meine LEDs spielen verrückt. Selbst der simple Strandtest funktioniert nicht. 
Alle LEDs blinken und wechseln völlig chaotisch ihre Farben. Manche sind heller andere sind dunkler. Ich hab allerdings kein Flackern. 
Ich habe richtig verlötet (Bernd_wiring) und ein 8A Netzteil. Keine Buttons bisher. Soundcard hab ich deaktiviert. 
Ich kann heute abend gerne ein Bild nachreichen. 
Ich bin momentan ein bisschen iritiert da ich einen mechanischen/elektrischen Fehler eigentlich ausschließe, die Software ja aber keine Atompyhsik ist und ich nichtmal das Testbeispiel hinbekomme. 
Kann mein dickes Netzteil das sehr nah am Raspi und den LEDs liegt so krass für Störungen sorgen? 

Grüße

Frosch

phenze, 2018-12-13:
Da ist wahrscheinlich eine LED irgendwo im Streifen hinüber.
Oder du hast ein kalte Lötstelle.
Das ist mir auch schon sehr oft passiert leider.

Hast du den Strandtest so umgeschrieben dass alle 114 LEDs verwendet werden ?
Dort würde ich auch alle Tests bis auf einen ColorWipe auskommentieren. 
So wird immer wieder nacheinander jede einzelne LED durchgeschalten und du kannst die fehlerhafte sehr leicht erkennen.
Ich hab dir in den Anhang mal eine modifizierte Version gepackt. Hab diese aber nicht getestet. Das war jetzt nur schnell aus dem Kopf.
[strandtest.py.zip](https://github.com/bk1285/rpi_wordclock/files/2675390/strandtest.py.zip)

Wenn das alles nichts hilft sieh die mal den Multistrandtest an.
Da gibt es einen zusätzlichen Parameter für die Konfiguration des LED Stripes (LED_1_STRIP).
Der Standardwert hierfür ist ws.WS2811_STRIP_GRB.
Dafür gibt es mehrere Konstanten (ws2811.h). Probier da einfach mal alle durch.

Hier noch die Links

- https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py
- https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/multistrandtest.py
- https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.h

phenze, 2018-12-13:
Hast du den Datenkanal auch gemessen ? oder nur +/-

Vielleicht brauchst du auch einen Level Shifter beim Zero. 

Siehe den anderen Thread 
https://github.com/bk1285/rpi_wordclock/issues/102

MaxZimmermann99, 2018-12-14:
Es funktioniert. Ich musste tatsächlich nur den Pi mit der LED Stromversorung zusammen erden. Wird irgendwo ja auch beschrieben. Hatte den Pi bisher über USB am Netz hängen. Jetzt funkioniert es sauber. Danke für die Hilfe und das Mitdenken!

closed 2018-12-14 by MaxZimmermann99