bk1285/rpi_wordclock #107: Hardware recommendations

phenze, 2018-12-13:
Since a few people(including me) have several problems with their LED stripes can anybody recommend any LED Stripe which is working without trouble ?

Perhaps we can update the hardware list in the documentation with some points you have to pay attention when buying the hardware ?

It is kind frustrating when building the clock with cheap/broken LED Stripes and then searching errors for weeks.

phenze, 2018-12-14:
Hello @FrankX0 ,
thank you for your input! (and for the great new Web Interface by the way !!)

Regarding your points i have the following input:

1. I will try this. Thank you. Is it complicated to connect the level shifter ?
2. Im wondering about that point. When you use the power supply you mentioned you have not enough power to get full brightness. The Strip you have mentioned have a Power Consumption of 18W/m (i guess thats for 60led/m). So our strips will have 9W/m. This means we need about 7A to get the full power ( 9W / 30 led = 0,3 W/led --> 114*0,3 / 5 = 6,84A). Do you have full brightness with your power supply ? 
3. Do you have any recommendations for "soldering noobs". Is there any easy way to connect the strips ?
4. Don't know what that means ? Sorry.

kimthostrup, 2018-12-14:
HI @phenze 
For point number 3, I have considered using these : https://www.ebay.com/itm/232374294719

phenze, 2018-12-14:
@kimthostrup that's great ! 

Thank you !

MaxZimmermann99, 2018-12-14:
Setup which works for me right now:

LED:
https://www.amazon.de/gp/product/B07439RXD3/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
PI:
Raspberry Pi Zero
Power:
https://www.amazon.de/gp/product/B00MWQF08C/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

Pi powered with 5V from LED power supply (did not work with Pi on USB)
No levelshifter
No buttons

Working fine, power supply only getting warm with all LEDs turned on.

ruedli, 2021-05-03:
Regarding the power supply: I have built 3 clocks where I for each use an old notebook power supply (23V) and I use 5V step-down convertors to step down to 5V. I use minimal 5: one for the Pi and 4 for 4 groups of LEDS. The ground is common for all step-down convertors. All 3 clocks work. Advantages:
- No voltage loss over the cable of the otherwise 5V power supply, when you loose voltage of the 23V, it is compensated by the subsequent step-down convertor.
- The step-downs are quite close to the LEDS, so less power loss with the 5V circuitry.
- No common 5V between Pi and LEDS
- So far no interference, the clocks are running for many years now, I am building the 4th one now.

Because I also use servos and motors in my clocks, they use different step-downs set to 6V.

The build deviates from the current description, but so far my experience is quite good. Anyone else gone this route?