bk1285/rpi_wordclock #108: Plugin to test strip for errors

phenze, 2018-12-14:
Hi @bk1285 ,

i have implemented my strandtest color wipe test script i am using for led strip error tests as a plugin.

https://github.com/phenze/rpi_wordclock/commit/8dbc07e7877c7b620ae146143325d8602dd9a4b4

Can you remerge it into your repo ?

bk1285, 2018-12-16:
Hey @phenze,

that's an excellent idea to support during the installation with some testing code.

However, I fear, that someone without a working installation wouldn't be able to select the according plugin.

How about transforming that into a "standalone"-script?

Best, 
Bernd

phenze, 2018-12-17:
Hi @bk1285 ,

[strandtest.py.zip](https://github.com/bk1285/rpi_wordclock/files/2684764/strandtest.py.zip)

But keep in mind that that errors could also appear after installation. That's why im developed the plugin.