bk1285/rpi_wordclock #110: Touching the Raspi crazy effects

MaxZimmermann99, 2018-12-23:
Hey guys if i touch my rasp zero with my finger it makes random stuff like changing the colour or starting random plugins. 
I have no buttons and my config says "gpio_high". 

Sometimes its even enough to touch the wood of the frame :) Not sure whats going on. My data cabel is soldered to the raspi so its not a buggy jumper cable...

Merry Chrismas everyone!

bk1285, 2018-12-23:
Hey @MaxZimmermann99,

did you setup one common ground? Raspberry + LEDs?

Best,
Bernd

MaxZimmermann99, 2019-01-19:
Sorry took me some days to get back here.

I have a common ground for the pi and the LEDs. Without this it wasnt working at all. https://github.com/bk1285/rpi_wordclock/issues/106

I really dont know what causes these effects. I'm gonna keep on reporting. Maybe the next clock is gonna be different :)

bhunziker, 2019-03-12:
Is it different if you pull the inputs to ground? It might be some cross-talk from the LED output

MaxZimmermann99, 2019-05-16:
Hey again everyone. I started over again. New Clock, new Raspberry Zero W, new 5V power and the same problem. 

I dont use any buttons and i tried gpio_high and gpio_low. I also tried to use a groud pin for all 3 buttons in the config but the clock won't start this way. Same with some random non existend pin numbers. 

Not sure how i am the only one with this problem as i'm using the very basic setup....
And not sure if anyone is still reading this post :)

Should i try to connect all 3 button pins with ground and use gpio_high?

Greetings
Max

P.S. Yes my drawing sucks but it should make clear how my setup looks like. (no buttons, no sensor, nothing else...)

[image]

bhunziker, 2019-05-25:
The only thing I could recommend you is what you proposed: all non used pins on ground.
btw: have you tried it with a very clean installation from github? Just in case it might be a software problem.

Greetings
Bernhard

Patrik42, 2019-06-30:
Hi All,
same problem at my end. 
I have got common ground, RPI 3b.

Additionally, the startscript via crontab does not work. 

The log shows 

  File "/home/pi/rpi_wordclock/wordclock.py", line 167, in <module>
    word_clock.startup()
    self.wcd.showText("IP: " + netifaces.ifaddresses(interface)[2][0]['addr'])
KeyError * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
:  * Environment: production
2   WARNING: Do not use the development server in a production environment.

   Use a production WSGI server instead.

Do I need to change the environment of Flask?!

SvenFackert, 2019-12-07:
For everyone who stumbles across this problem in the future (no hardware buttons), I have a very simple solution for you. I had the same problem for one of my builds that had no buttons. In these cases touching the pins of the Raspi will eventually trigger the button functions.

The simplest solution in my opinion is to just disable the event listeners for the buttons in the wordclock_interfaces/gpio_interface.py file. Open the file e.g. in nano (sudo nano rpi_wordclock/wordclock_interfaces/gpio_interface.py) and comment out the following lines (by inserting the hash/number sign at the beginning of each line):

`#        GPIO.add_event_detect(self.button_left,
#                              self.polarity,
#                              callback=lambda channel: self._left(),
#                              bouncetime=100)
#        GPIO.add_event_detect(self.button_return,
#                              self.polarity,
#                              callback=lambda channel: self._return(),
#                              bouncetime=100)
#        GPIO.add_event_detect(self.button_right,
#                              self.polarity,
#                              callback=lambda channel: self._right(),
#                              bouncetime=100)`

Finally CTRL + O, ENTER to save your changes and CTRL + X to quit nano.

Maybe this could also be configured in the config file in future versions. E.g. a property "disable_hardware_buttons".

mrksngl, 2019-12-10:
The hardware fix would be to hardwire the corresponding gpios to ground (or VCC, depending on the polarity setting) :)

But yes, we should also give it a property, as it seems there is a use case for having no buttons.

bk1285, 2021-07-24:
Reopen, if still unresolved.

closed 2021-07-24 by bk1285