bk1285/rpi_wordclock #116: Set plugin from script

MarioVicc, 2019-03-01:
Hi

First of all, let me thank you all for this great project!

I’m looking for the best way to switch to a specific plugin from a (python) script and hope you guys can help me out.

(I already have a script that determines if I’m present by checking whether my phone is in Bluetooth range and I’d like to turn the wordclock on and off accordingly)

Thanks
Mario

MarioVicc, 2019-03-04:
To answer my own question for future user:

You can use the included RESTful API to interact with the wordclock.
Issuing requests to a web server seems a bit cumbersome to me, but it works:

`import` requests

headers = {
    'accept': 'application/json',
    'Content-Type': 'application/json',
}

time_default = '{  "name": "time_default"}'

response = requests.post('http://127.0.0.1/api/plugin', headers=headers, `data=time_default)`

you'll have to install requests first:

`pip` install requests`

MarioVicc, 2019-03-04:
And here the actual script I’m running based off prb3333’s work over at https://www.instructables.com/id/Raspberry-Pi-Bluetooth-InOut-Board-or-Whos-Hom/
Check out that post to see how to set it up!

```
import bluetooth
import time
import os
import requests

headers = {
    'accept': 'application/json',
    'Content-Type': 'application/json',
}

time_default = '{  "name": "time_default"}'
led_off = '{  "name": "leds_off"}'

time.sleep(120)

count = 0

result = bluetooth.lookup_name('AA:BB:CC:DD:EE:FF', timeout=5)
if (result != None):
	response = requests.post('http://127.0.0.1/api/plugin', headers=headers, data=time_default)
	user_is_present = 1
else:
	response = requests.post('http://127.0.0.1/api/plugin', headers=headers, data=led_off)
	user_is_present = 0

while True:
    result = bluetooth.lookup_name('AA:BB:CC:DD:EE:FF', timeout=5)
    if (result != None):
		count = 0
		if (user_is_present == 0):
			response = requests.post('http://127.0.0.1/api/plugin', headers=headers, data=time_default)
			user_is_present = 1
    else:
		if (user_is_present == 1):
			count = count + 1
			if (count > 5):
				response = requests.post('http://127.0.0.1/api/plugin', headers=headers, data=led_off)
				user_is_present = 0
				count = 0

    time.sleep(20)
```

bk1285, 2019-08-01:
Nice! :)

closed 2019-08-01 by bk1285