bk1285/rpi_wordclock #117: Color Wheel on web interface not working with white

bhunziker, 2019-03-12:
Hi all

First of all, let me thank you all for this great project!

I realized that when setting the default color to WHITE in the time_default plugin (line 47-49) the color wheel does not work any more. My workaround is to change the color before handing it to the web, but this is not nice. Any other ideas how to fix it?

bk1285, 2019-03-12:
Hi @bhunziker 

I just had the chance to check this, but cannot reproduce the issue, using latest develop, master.

Is it possibly browser specific? Can you test with another browser/device/...

Best,
Bernd

bhunziker, 2019-03-13:
Hi Bernd

Thanks a lot for your fast and useful reply. I checked it it with the latest master and realized it works. So I have to do some homework to find out what's different.
The browser (chrome, IE, safary) has no influence.

closed 2019-03-13 by bhunziker