bk1285/rpi_wordclock #118: No more soldering

viezefreddie91, 2019-03-20:
After destroying my LED strip twice because of my genius soldering, i found these “double connectors”. Just wanna share..

https://m.nl.aliexpress.com/item/32870470660.html?pid=808_0009_0101&spm=a2g0n.search-amp.list.32870470660&aff_trace_key=e7f289f08a6541f2b8f323627bc35e62-1548465616450-07303-cWkwJnjS&aff_platform=msite&m_page_id=470amp-UeAh5rejSePDYH1n4dqQnw1553108817837

bk1285, 2019-03-20:
Hey @viezefreddie91,

thanks for sharing! I'll add them to the list of components... :)

Best,
Bernd

bk1285, 2019-03-27:
Here we go... https://github.com/bk1285/rpi_wordclock/commit/ebfa30599b06184313b6d5857e15de421299b8d7

closed 2019-03-27 by bk1285

gielk, 2022-10-17:
@viezefreddie91 The link no longer works, which one did you buy? Do you maybe have the name you found it with on AliExpress?

sezijlema, 2023-01-22:
I have not used or tried them but I think he means e.g. these: https://www.aliexpress.com/item/1005001521762588.html
Depending on your LED strip, you will need a certain version.

For the Dutchies among us, here's a local option as well: https://www.tinytronics.nl/shop/nl/verlichting/led-strips/accessoires/led-strip-connector-15cm-3p-10mm-ws2811-ws2812b

If these links die as well, I found them by searching for "LED strip Connector (cable)".

gielk, 2023-01-22:
> I have not used or tried them but I think he means e.g. these: https://www.aliexpress.com/item/1005001521762588.html Depending on your LED strip, you will need a certain version.
>
> For the Dutchies [...]

thanks 

dankjewel! Heb jij er al een gebouwd of ben je het van plan?