bk1285/rpi_wordclock #121: Development Mode on Debian Linux fails to run

coolbikerdad, 2019-06-08:
Hi. 

I am trying to understand the code before building my word clock (parts on order), and would like to build and run wordclock.py on Linux with GTK in development mode. Unfortunately, I cannot get the program to run. I have set an appropriate configuration file, fixed up font path issues, but the main error now seems to be that the colour values for GTK appear to be integers rather than tuples. In GTKstrip.py, line 105 I get an exception thrown: 
AttributeError: 'int' object has no attribute 'r'
and the list in self.colors is a list of integers, not Color objects.
Is there a Linux/GTK setup I can use to get the program running? I am on latest x86 Debian Stretch. I wonder if the GTK libraries have moved on since the program was written? Hads development mode been tested recently? Love what the software will do!

regards
Neil Matthew

bk1285, 2019-08-01:
Dear Neil,

just verified, that it works on my PC, running Ubuntu 18.04.2 LTS.
@coolbikerdad: With "development mode", you're referring to the following configuration within rpi_wordclock/wordclock_config/wordclock_config.cfg, right?
```
[wordclock]
developer_mode = True
```
@mrksngl: Any idea?

Best,
Bernd

andredatele, 2019-08-07:
I get this same error on both raspbian and also Ubuntu 18.04.2 LTS:

[image]

bk1285, 2021-07-24:
Reopen, if unresolved.

closed 2021-07-24 by bk1285