bk1285/rpi_wordclock #122: Access the wordclock via webinterface

echo1483, 2019-06-26:
Access the wordclock via webinterface
Hallo leider nur in Deutsch, Sorry
Die Uhr läuft soweit super mit einem raspberry pi zero w.
Besten Dank für die Arbeit und Bereitstellung des Programms.
Nur das mit dem WEB-Interface kriege ich nicht hin.
Wenn ich im browser meine IP eintrage bekomme ich diese Meldung.
Control Colors fa fa-question-circle The Raspberry Wordclock fa fa-link Project page
fa fa-link Contributors
fa fa-cogs API of this clock
fa fa-caret-left fa fa-circle fa fa-caret-right
Damit kann ich nichts anfangen, ich hoffe das mir jemand helfen kann.

Danke echo1483

PS. Übersetzung mit Google.

The clock is running so far great with a raspberry pi zero w.
Thank you for the work and delivery of the program.
Only with the WEB interface, I can not get there.
When I enter my IP in the browser I get this message.

Control Colors fa fa-question-circle The Raspberry Wordclock fa fa-link Project page
fa fa-link Contributors
fa fa-cogs API of this clock
fa fa-caret-left fa fa-circle fa fa-caret-right

I can not do anything with that, I hope someone can help me.
Thanks echo1483

FrankX0, 2019-06-28:
It seems to me it is not able to load all resources.
Does your pi have internet access? Some of the resources need to be fetched externally.
If you are using Chrome, can you "inspect" the web-page (crtl-shift-i) and check for errors (top right corner)?

BTW: you can respond in German.

echo1483, 2019-06-28:
Ist auch für mich einfacher, Danke.
Die Uhr wird am Router mit IP Adresse angemeldet. Eigentlich ist alles in Ordnung.
Nur wenn ich die Uhr über den Bowser ansprechen möchte bekomme ich folgende Antwort.
Control Colors fa fa-question-circle The Raspberry Wordclock fa fa-link Project page
fa fa-link Contributors
fa fa-cogs API of this clock
fa fa-caret-left fa fa-circle fa fa-caret-right
Dann ist auch schon Ende.

Mfg echo1483

FrankX0, 2019-06-28:
Clear.
Do you use Chrome as browser?
If so, can you "inspect" the web-page (crtl-shift-i) and check for errors (top right corner)?

What you are seeing in your browser is part of the html code.

bk1285, 2019-06-28:
Hey! Indeed, this might be a browser issue. Try several different, such as firefox, chrome, ... on differebt devices: Laptop, Smartphone,  ...

Best,
Bernd

echo1483, 2019-06-29:
Danke für die Hilfe.
Ich habe mal verschiedene Browser ausprobiert und mein Tablet immer das gleiche. Als Info kam immer die Seite ist nicht sicher. Ich habe darauf alle Berechtigungen für diese Adresse erteilt.

Aufruf der Wordclock.
[image]

Wenn ich auf API of this Clock klicke erhalte ich diese Meldung.
[image]

Firefox ist mein Standardbrowser.

MfG echo 1483

FrankX0, 2019-06-29:
You seem to have no access to o.a. stylesheets (first screen).
The second screen is strange, but suggests you are not running the latest code. Can you update?

In Firefox, can you right-click on the page and click **Inspect Element**? Does it show errors?

echo1483, 2019-06-29:
Ich habe die aktuelle Version der Wordclock und der Browser.
Im Firefox und Elements bekomme ich folgende Info:

[image]

[image]

Leider kann ich damit nichts anfangen. 

Gruß echo1483 und danke für die Hilfe

FrankX0, 2019-06-29:
Well, there are definitely issues with fetching (local) resources, which is most likely the cause of what you are seeing:
[image]
But I do not understand why this happens in your case.

Can you try to load one of the mentioned files by hand?
So: `192.168.178.46/static/js/vue-color-picker.js`
This should result in the js file to be displayed.

echo1483, 2019-06-29:
Ich habe folgende Adresse "http://192.168.178.46/static/js/vue-color-picker.jsim" im Firefox eingegeben, ich hoffe das war richtig. Dannach bekomme ich dieses Meldung:

[image]
Gruß echo1483

echo1483, 2019-06-30:
Liegt der Fehler an der Raspbian OS-Version oder fehlender Module ?
Ich habe mal meine Version ausgelesen:

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

FrankX0, 2019-06-30:
I think this is the problem: you have Apache running.
They are both listening to port 80. I don't think this will work.

Please stop/uninstall Apache or use a different port for the web interface.
To (temporarily) stop Apache: `sudo systemctl stop apache2.service`

Hope this helps.

echo1483, 2019-06-30:
Wenn ich den Service stope bekomme ich keine Verbindung mehr.
Ich habe es mal mit dieser Version 2019-06-20-raspbian-buster-full versucht, läuft aber auch nicht.
Welche Raspbian.... Version ist denn am sichersten ??

Danke echo 1483

FrankX0, 2019-06-30:
Stretch is fine (running myself), no need to update.
If you do not get any response, then the Python webserver (flask) is not running (@bk1285 correct?).

When you run `sudo python wordclock.py` from a terminal, you should see at the end:
` * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)`
Do you see this, or any errors?

echo1483, 2019-06-30:
Den Befehl habe ich ausgeführt. Ich erhalte eine Menge Informationen, aber "Running on http://0.0.0.0:80/" ist nicht dabei. Ich hoffe das ich Sie nicht nerve mit mein Problem.

gero1483

pi@raspberrypi:~/rpi_wordclock $ sudo python wordclock.py
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
  Setting language to german.
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be                                       imported.
Imported plugin 0: "rainbow".
Imported plugin 1: "matrix".
Imported plugin 2: "tetris".
Imported plugin 3: "time_as_words_german".
Imported plugin 4: "time_as_words_dutch".
  Assumes no temperature sensor to be attached.
Imported plugin 5: "weather_forecast".
Imported plugin 6: "time_matrix".
Imported plugin 7: "leds_off".
  Selected "time_default" as default plugin
Imported plugin 8: "time_default".
Imported plugin 9: "sunrise".
  INFO: No activate-flag set for plugin time_in_seconds within the config-file.                                       Will be imported.
Imported plugin 10: "time_in_seconds".
Imported plugin 11: "ip_address".
Imported plugin 12: "shutdown".
Imported plugin 13: "feed_parser".
Imported plugin 14: "restart".
Unhandled exception in thread started by <bound method web_interface.threaded_ap                                      p of <wordclock_interfaces.web_interface.web_interface instance at 0xb5c74800>>
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 61,                                       in threaded_app
    self.app.run(host='0.0.0.0', port=port)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 841, in run
    run_simple(host, port, self, **options)
  File "/usr/lib/python2.7/dist-packages/werkzeug/serving.py", line 708, in run_                                      simple
    inner()
  File "/usr/lib/python2.7/dist-packages/werkzeug/serving.py", line 670, in inne                                      r
    fd=fd)
  File "/usr/lib/python2.7/dist-packages/werkzeug/serving.py", line 564, in make                                      _server
    passthrough_errors, ssl_context, fd=fd)
  File "/usr/lib/python2.7/dist-packages/werkzeug/serving.py", line 476, in __in                                      it__
    HTTPServer.__init__(self, (host, int(port)), handler)
  File "/usr/lib/python2.7/SocketServer.py", line 417, in __init__
    self.server_bind()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 108, in server_bind
    SocketServer.TCPServer.server_bind(self)
  File "/usr/lib/python2.7/SocketServer.py", line 431, in server_bind
    self.socket.bind(self.server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
Running plugin time_default.

FrankX0, 2019-06-30:
The Python webserver is indeed not running, due to:
`socket.error: [Errno 98] Address already in use`

So you either remove Apache (if you do not need a generic webserver), or you use a different port for the webinterface: https://github.com/bk1285/rpi_wordclock/blob/52057f4b1e192b27633e483fc0165d842b93805a/wordclock_interfaces/web_interface.py#L60
Change 80 to something else, e.g. 8080.
Of course you then need to access the webinterface by specifying the port, as it is different from the default (80): http://192.168.178.46:8080

echo1483, 2019-06-30:
Vielen Vielen Dank, Du bist mein Held.
Durch die Änderung der Zeile 60 und den Port auf 8080 war der Fehler behoben.
Also vielen Dank Dietmar aus Kempen.
 
Hier zwei meiner Uhren.

[image]

Die Wordclock habe ich in 50 cm und 25 cm gebaut. Die Zeitmatrix habe ich in 0.8mm Flugsperrholz gefräst.

[image]

FrankX0, 2019-06-30:
Gern geschehen, Dietmar!
Nice clocks!

echo1483, 2019-07-29:
Hallo FrankX0
seit einigen Tagen kann ich die Wordclock nicht mehr über das Webinterface erreichen. Ich habe nichts verändert, bis dato lief alles ohne Probleme. Als erstes habe ich beim RPI ein Update und Upgrade durchgeführt, half aber nichts. Wenn ich die Uhr aufrufe erhalte ich nur eine leere Seite. Im Log sind diese Meldungen:

Error: "Vuetify is not properly initialized, see https://vuetifyjs.com/getting-started/quick-start#bootstrapping-the-vuetify-object"
    beforeCreate VApp.ts:39
    VueJS 20
    <anonymous> scripts.js:2
vue.min.js:6:11752
[image]

Ich hoffe Du kannst mir mal wieder helfen.
Vielen Dank, Dietmar

FrankX0, 2019-07-29:
Hi Dietmar,
Thanks for reporting. This looks like a "real" issue: I also do not have the webinterface when trying from the camping site in Austria (currently on holiday).
So I can only have a look when I am back home.
It looks like some referenced resources have been updated, causing a backwards compatibility issue. I probably can have a look coming weekend.

echo1483, 2019-07-29:
Hallo,
danke für die schnelle Antwort, ich hatte mir schon Sorgen gemacht.
Also schönen Urlaub und alles Gute.
Gruß Dietmar

koko0815, 2019-07-31:
Hi folks,

thanks for that great project. I've done this with my students. At the moment we have the same issue with the webserver. It would be great to here what we can do to fix it.

Thanks for that

Best regard
Philipp

FrankX0, 2019-08-01:
@echo1483 @koko0815 I just created a pull request #128 to fix the found issue.

bk1285, 2019-08-01:
Dear @koko0815,

it's great to hear, that you do this project with students!?! Do you have any feedback, how to improve the learning effect/educational aspects? ...taking tools/documentation/etc. into consideration.

Best,
Bernd

echo1483, 2019-08-02:
Danke FrankX0 und Bernd für die schnelle Hilfe, die Wordclock ist ein super Projekt. Ich bin schon auf weitere Module gespannt.

Thanks FrankX0 and Bernd for the quick help, the wordclock is a great project. I'm looking forward to more modules.

Best Dietmar

closed 2019-09-15 by bk1285