bk1285/rpi_wordclock #126: Blank web interface

kendersec, 2019-07-31:
I've had to reboot the wordclock recently and I've noticed that for at least a couple of weeks the web interface appears to be blank. I get a response from the server but the content doesn't render at all. I believe the HTML content is there but the page is still blank.

My current suspicion is that the vue or vuetify scripts have been updated and the code in the wordclock repository is no longer compatible with the newer versions. I tried downgrading the scripts in the HTML to some order version and got some of the UI rendering but only partially, the colour wheel would still not show up and the UI would be partly broken.

I have no knowledge of vue or vuetify but happy to help in what I can.

FrankX0, 2019-08-01:
I just created a pull request #128 to fix the found issue.

bk1285, 2019-08-01:
Thanks for your PR, @FrankX0 
@kendersec can you verify the fix to be working for you (using latest master-branch)? If so, I'll close this issue.

Best,
Bernd

kendersec, 2019-08-01:
Awesome guys, it's all back to normal now. Thanks a lot @FrankX0 and @bk1285

closed 2019-08-01 by bk1285