bk1285/rpi_wordclock #132: Grid not centered vertically

SvenFackert, 2019-09-08:
I just wanted to point out, that the character grid is not perfectly centered vertically. It is a bit too far to the bottom. Maybe the problem origins from the vertical alignment of font characters inside the SVG text elements. But I didn't have any time to dig deeper into that yet. 

A quick and dirty fix that I just used would be to just offset each character by 4 pixels which will output an almost perfectly centered grid.

create_layout.py (only added the -4)
```
...
if mode == 'stencil':
    # Write only characters
    text_layout.add(layout.text((content[y].decode('utf-8')[x]),
                                insert=(coords[0], coords[1] + (float(font_size) / 2.0) - 4)))
...
```

Big thanks to all of you guys working on this awesome project and especially to @bk1285 !

bk1285, 2019-09-15:
Hi @SvenFackert 

thanks for reporting. Can you detail, which viewer you were using? I'll check, if I can reproduce that.

Best,
Bernd

SvenFackert, 2019-09-15:
Hi Bernd, 

I checked it using Google Chrome as well as Adobe Illustrator. The offset appeared in both of them.

Kind regards,
Sven