bk1285/rpi_wordclock #133: Issue running wordclock.py

basholtman, 2019-09-14:
Can anyone point me in the right direction, the strandtest.py is running fine al leds work.
But when I try to run wordclock.py i'm getting the following messages.
[image]

I've no buttons attached to my clock and set gpio already to high, also already disabled my soundcard and running on a raspberry pi zero w.
I'm quit new to programming so would need clear directions.

bk1285, 2019-09-15:
Hey @basholtman,

this seems to occur due to missing fonts. Try to rerun the commands below, as pointed out in the documentation: https://rpi-wordclock.readthedocs.io/en/master/doc_software_setup.html

```
sudo apt-get install python-pip python-scipy scons git swig ttf-freefont
sudo pip install pytz astral feedparser pillow svgwrite freetype-py netifaces monotonic flask-restplus

```

... where `ttf-freefont` and `freetype-py` should be the relevant packages to fix the error above...

Best,
Bernd

basholtman, 2019-09-16:
Hey Bernd thank you for the reply,

[image]
I have all the other packages and cant install ttf-freefont, am I missing something?

plotaBot, 2019-09-16:
Maybe it has to deal withe fact that you are using duster.
Using stretch every dependencies needed installed fine.
If you want to try stretch you can download it [here](http://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/).

basholtman, 2019-09-17:
That's maybe something for when i'm further down the road with the issue plotabot.

[image]
[image]

Managed to install the mscorefonts, but still no luck on the ttf-freefont (Should work with duster according to Debian) and running the wordclock.py 
Any help would be greatly appreciated

basholtman, 2019-09-21:
@plotaBot indeed ttf-freefonts does not install buster lite, it installs fine in stretch lite. Lets see if I can get the clock running on stretch.

basholtman, 2019-09-21:
issue closed: this project runs fine on Raspbian stretch lite, but doesn't Raspbian buster lite due to the fact ttf-freefonts can't be installed. 
@bk1285 Might want to add that in the troubleshooting section or at main build page @ software setup

closed 2019-09-21 by basholtman