bk1285/rpi_wordclock #135: Wired wrong direction, how to fix

O365Tycho, 2019-11-05:
I didn't pay attention and wired my clock in a different direction. As you can see in the attached image i started with the minute leds in the top left corner, followed by bottom left corner and then went up,down,up,down etc with the led strings. This results in the leds being vertically mirrored (don't know how else to describe it).  

What is the best way to fix this? I found wordclock_plugins/time_default/time_dutch.py and think i can probably make the time work by modifying the ranges in this file, but that probably doesn't resolve the issue with other plugins.

[image]

ps. Did a git pull today and am now getting errors on the web interface. Should i create an issue for it or is it something i'm doing wrong?
[image]

HeerNMeester, 2019-11-06:
Hi Tycho,

I see that rewiring is not really an easy option anymore.
The wiring of the LEDS is configured in this file:
https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wiring.py
You can create your own wiringscheme here and use that in the config file.

O365Tycho, 2019-11-07:
Thanks for the link to the wiring scheme. I've added my own wiring class with the following formulas in the getStripIndexFrom2D(self,x,y)

`if x % 2 == 0:
   pos = (self.WCA_WIDTH -x) * self.WCA_HEIGHT -y + 1
else:
   pos = (self.WCA_WIDTH -x - 1) * self.WCA_HEIGHT + y + 2`

closed 2019-11-07 by O365Tycho