bk1285/rpi_wordclock #136: Font suitable for laser cutting

O365Tycho, 2019-11-06:
I like the new WCFont that has been recently commited, but i think a font suitable for laser cutting is needed with modified letters like A,B,D,O,Q,R,P . Would be nice if such a font is included by default to use with the create_layout.py script. 

See for example the one @HeerNMeester meester used in his project:
[https://user-images.githubusercontent.com/22868008/29250896-279badb2-804b-11e7-8805-2514a243e2e9.jpg]

HeerNMeester, 2019-11-06:
Hi Tycho,
There are 3 options for getting the right font:
1. Search the internet ;)
2. Create you own
3. Use this website to create your own template (Use the Stencil DIN Q2 font). 
[https://simongolms.github.io/QLOCKGENERATOR/#/home](url)

@bk1285 Maybe a good idea to link this site on the ReadTheDocs?

O365Tycho, 2019-11-06:
Nice site! Guess that beats using the included python script. Definitely something that should make it to the docs.

closed 2020-01-01 by bk1285