bk1285/rpi_wordclock #138: Question - Translate - New Stencil

kimthostrup, 2019-11-13:
Gents,

I would like to translate the existing stencils to Danish, any idea how to go around this task ?

Thanks in advance.
/Kim

plotaBot, 2019-11-13:
Use this website: https://files.qlocktwo.com/Q2/check.html
btw: This is from the original designers. Pretty sweet I have to admit.
UR welcome;)
@bk1285 please add this to your doku. It is very helpful for future implementaions of new languages.

kimthostrup, 2019-11-13:
> Use this website: https://files.qlocktwo.com/Q2/check.html
> btw: This is from the original designers. Pretty sweet I have to admit.
> UR welcome;)
> @bk1285 please add this to your doku. It is very helpful for future implementaions of new languages.

Thank you for your feedback.

Sorry for my ignorance,  but how will this help in getting the logic behind the rpi wordclock to light up the correct led's to "spell" the led words that of course is different depending on the language ?

plotaBot, 2019-11-13:
Sorry the site is in german xD
If you choose "DK Dänisch"  under "Sprachen" you are able to manually switch trough each hour (with ++) and minute (with +) on the website. I have preselcted the settings you need [`here`](https://files.qlocktwo.com/Q2/check.html?color=BLACK%20ICE%20TEA&controls=false&hours=0&language=DK&minutes=0&size=450):
[image]

Also take a look at a language pack and make yourself familiar with the logic behind its creation. You can use the [`english`](https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/time_english.py) one as a template.

kimthostrup, 2019-11-13:
> Sorry the site is in german xD
> If you choose "DK Dänisch" under "Sprachen" you are able to manually switch trough each hour (with ++) and minute (with +) on the website. I have preselcted the [...]

Perfect thanks :-)

bk1285, 2019-11-13:
Dear @kimthostrup,

good to hear, that you want to add danish language support 👏 

The (currently undocumented) procedure is the following:
* Research a layout (e.g. using @plotaBot's link: https://files.qlocktwo.com/Q2/check.html)
* Add this layout to your config-file at the bottom (see examplary file [here](https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_config/wordclock_config.example.cfg), search for `language_options`)
* Run python create_layout.py -a within the wordclocks main directory to create svg/png files of the new language in the folder wordclock_layouts (such as: https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_layouts)
* Add the corresponding logic to display the correct leds in a new file [here](https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_plugins/time_default)
* Add your new language to your local [wordclock_tools/wordclock_display.py](https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wordclock_display.py)
  * add the corresponding import at the top of the document
  * add the corresponding if-clause [here](https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wordclock_display.py#L92)
* Update the section `[wordclock_display] language = german` within your config-file.
* Restart your wordclock and check, if everything works as expected
* Finally, feel free to complete/improve this post, since this might become the documentation for future language integrations ;)

Feel free to comment, in case you need further support.

Best,
Bernd

kimthostrup, 2019-11-13:
> Dear @kimthostrup,
>
> good to hear, that you want to add danish language support 👏
>
> The (currently undocumented) procedure is the following:
>
> * Research a layout (e.g. using @plotaBot's link: [...]

Hi Bernd,

Thank you for the information above, as soon as I have my (second) rpi_wordclock build completed, I will give this a shot.

bk1285, 2021-07-24:
Please reopen, if still unresolved.

closed 2021-07-24 by bk1285