bk1285/rpi_wordclock #139: Brightness sensor

SvenFackert, 2019-11-14:
Hey folks,

I've been building a few of these WordClocks now and am very grateful for the really awesome work that everyone contributed here. Especially the documentation on how to build the clock is very exhaustive and easy to understand (even for ppl without any experience in hardware).

What I was missing up to now was the support for a brightness sensor that automatically dims the clock according to the environmental brightness, so I've been building one by myself from just a photoresistor, capacitor and a few lines of code that counts the time until the capacitor is charged and then adjusts the brightness of the word clock.

I was wondering if you guys are interested in 1) a documentation on how to build the sensor and 2) the code for it.

Kind regards,
Sven

kimthostrup, 2019-11-14:
> Hey folks,
>
> I've been building a few of these WordClocks now and am very grateful for the really awesome work that everyone contributed here. Especially the documentation on how to build the clock [...]

Of course please share :-)

/Kim

phenze, 2019-11-14:
@SvenFackert take a look here: 

https://github.com/bk1285/rpi_wordclock/issues/27

I have a brightness sensor until day one of my clock. 
Would be great if we can find a solution for everyone.

SvenFackert, 2019-11-14:
Ah.. I see. There has already been a list of requirements for an integration of a brightness sensor: https://github.com/bk1285/rpi_wordclock/issues/27#issuecomment-450756826

My solution doesn't meet those requirements yet. Also it is somehow dependent on the type of sensor I am using. 

I suggest that we maybe create an independent auto_brightness interface, which is responsible for providing an API to enable/disable automatic brightness (as well as settings in the config file). Furthermore we could move the sensor specific readings/code to separate files which should all provide an interface for reading the current lumen value or sth. like that?

bk1285, 2019-11-24:
Sounds good, @SvenFackert. Let's detail on that a little further. Maybe start over with sharing your implementation (e.g. via a fork + branch in your own repo)..

Best,
Bernd

spintike, 2020-03-30:
Hi guys!

@bk1285 your WordClock software still is amazing, I am loving it. My own clock is running without a flaw since approx. 1 year.
Now I want to built another one for my family and so far everything is set up. However at the time I ordered the parts I was only able to acquire a different light sensor (GH302 BH1750). Does somebody have experience with this kind of sensor and knows how to implement it into the current built?

Help would be greatly appreciated!

Sensor is this one: https://cmosgr.wordpress.com/2013/09/02/bh1750fvi-gy-302-light-sensor-module/

Cheers.

phenze, 2020-03-30:
That should no be different to my implementation.

Take a look here:
- https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L117
- That is the import of the Light sensor Library
- https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L131
- There we get the current value
- https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L154
- There we calculate a value for our wordclock

Step by Step:

- Get the maximum and minimum value coming from your sensor (you have to test this with the example python script). make this test when there is sunlight outside
- In my example the minimum is 0 (no light) and 100 (maximum light)
- Next you define the brigthness values of your clock which belong to the minimum/maximum value of the sensor. In my example this are 50 (i dont want to complete disable the leds when it is dark) and 255 (i want full power when outside is sunlight)
- Replace your values in the script Line 135 - 141
- Implement your logic to get the current value.

phenze, 2020-03-30:
That should no be different to my implementation. But you need a python library for that. Your example is C++.
Test this one:
https://gist.github.com/oskar456/95c66d564c58361ecf9f
What are your knowledge in python ? 

For changes in wordclock software take a look here:
- https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L117
- Here you have to do the import of the sensor library
https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L131
- Here you can define your default values
- https://github.com/phenze/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L154
- And here you must read your value.

Step by Step:

- Get the maximum and minimum value coming from your sensor (you have to test this with the example python script). make this test when there is sunlight outside
- In my example the minimum is 0 (no light) and 100 (maximum light)
- Make the import of your Sensor Library (Line 117)
- Next you define the brigthness values of your clock which belong to the minimum/maximum value of the sensor. In my example this are 50 (i dont want to complete disable the leds when it is dark) and 255 (i want full power when outside is sunlight)
- Replace your values in the script Line 135 - 141. 
- In Line 144 my sensor is initialized. For you this should look like (adopt this for python):
`Serial.begin(9600);
lightMeter.begin();`
- Read out the current value and calculate the brightness value for the clock (Line 156 - 159)
- in your case (of course then in python):
`uint16_t lux = lightMeter.readLightLevel();`

spintike, 2020-03-30:
> That should no be different to my implementation. But you need a python library for that. Your example is C++.
> Test this one:
> https://gist.github.com/oskar456/95c66d564c58361ecf9f
> What are your [...]

To answer your question regarding my python knowledge: "I am a small beginner" and even that seems exaggerated.
So far I was able to follow your manual and

- read out the minimum and maximum value of the sensor when in sunlight and darkness
- replace lines 135 to 141 with the values of my sensor and brightness

I assume that by _"Make the import of your Sensor Library (Line 117)"_ you mean to import the "bh1750.py" script from the first link? Because I do not understand completely how to achieve this...simply by replacing `import Adafruit_GPIO.I2C as I2C` with `import bh1750.py as I2C`? Or is it meant to replace the `control_on`& `control_off` values with the ones out of the BH1750.py script?

Unfortunately I do not understand the last two points regarding line 144 and 156-159.

phenze, 2020-04-01:
Okay without Python knowledge this could be hard.

A second approach you can try is just to replace the I2C Values.

For Your Sensor (i think):
- Adress : 0x23
- Power On : 0x01
- Value: 0x20 or 0x21 or 0x23

For more details and perhaps comparison between the python scripts take a look here:

https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/I2C.py

Feel free to ask further questions if you need information

spintike, 2020-04-07:
@phenze ok I got it working and the solution is so easy that I am bit ashamed...even with my modest python knowledge.

The solution is to replace the I2C device address in the plugin file (to the address 0x23 of the sensor I used):

https://github.com/phenze/rpi_wordclock/blob/d52ce76d5633a182cf8192a1f75848dbee45aff2/wordclock_plugins/time_default/plugin.py#L156

Thanks a lot for the inspiration @phenze!!