bk1285/rpi_wordclock #140: Blank Web Interface - Flask not rendering template?

coolbikerdad, 2019-11-23:
Hi. 
I have had to make a new installation of my word clock and I am not getting the web interface rendering. I get a blank page on visiting http://<ip_address of clock>, and "view source" shows me the contents of app.html as-is. Calls to the APIs give expected results though, so flask is running. Looks like Vue is not doing its thing. I have been running the clock beautifully since June (a power failure zapped my SD card), and a new build on Raspbian Stretch and Duster give the same results. Maybe something in a recent Raspbian update has broken. This is the case for a new installation just from the git sources, not my customised configuration. Has anyone else had problems with new installs lately?
Regards, Neil

FrankX0, 2019-11-24:
Hi Neil,
I can confirm that the web interface currently results in a blank screen.
And yes: Vue is not doing its thing.

It fails to load `iro.min.js` from `static/js` (which is not there).
Furthermore there is a reference error: `colorByTemp is not defined`

These changes have been made from end of September.
@bk1285 @plotaBot can you have a look?

bk1285, 2019-11-24:
Thanks for reporting, @coolbikerdad, @FrankX0.

The issue of the missing `iro.min.js` should be resolved. Please update to lastest master. The reference error however, I cannot reproduce. Please report, if the issue continues to exist...

Best,
Bernd

FrankX0, 2019-11-24:
Hi Bernd,

The web interface now renders correctly again (reference error is also gone now).
I however still see two errors when loading the page:

- `Failed to load resource: vue-color-picker.css`
But with the new color picker this css is not used anymore. Correct?
- `GET method not allowed: /api/color_temperature`
The clock API does not seem to support this method. Correct?

Regards,
Frank.

bk1285, 2021-07-24:
Please reopen, if still unresolved.