bk1285/rpi_wordclock #141: Time of day - sleep not working.

SteveThomson19, 2019-12-02:
Hi,  Relatively new programmer here but I've reached my limit and no further forward to fixing the problem.

It seems my nighttime sleeping isn't functioning (or being picked up by the main program).

In my wordclock config file everything else seems to be working.
Section of Config file.

```
[plugin_time_default]
activate = True
# animate new time with typewriter animation
typewriter = True
typewriter_speed = 10
# show time without prefix IT IS/ES IST/etc.
purist = False
# define sleep times when the display brightness is turned down [define any 
sleep_begin_hour = 22	# [0-23]
sleep_begin_minute = 00	# [0-59]
sleep_end_hour = 07	# [0-23]
sleep_end_minute = 00	# [0-59]
sleep_brightness = 10	# [0-255]

```
I have tried all combinations of the numbers e.g.  07, 7  etc to no avail.

Upon startup the following happens.
 ```
 Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to english.
Imported plugin 0: "leds_off".
Imported plugin 1: "feed_parser".
Imported plugin 2: "sunrise2".
Imported plugin 3: "restart".
Failed to import plugin time_matrix!
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be imported.
Imported plugin 4: "rainbow".
   No sleeping time set, display will stay bright 24/7
   No sleeping time set, display will stay bright 24/7.
  No sleep brightness set within the config-file. Defaulting to 5.
  Selected "time_default" as default plugin
Imported plugin 5: "time_default".
```

As you can see, it doesn't like the brightness setting (even though it's set) , but it doesn't recognise the sleep parameters which is my main problem.

Is there any further testing I can do to narrow down to what the issue is, I don't believe I've touched anything else ?   

Thanks in advance.

phenze, 2019-12-03:
Just remove the comments behind the variables.

sleep_begin_hour = 22
sleep_begin_minute = 00
sleep_end_hour = 07
sleep_end_minute = 00
sleep_brightness = 10

SteveThomson19, 2019-12-03:
Thank you,  tried that and it crashes.

Cloned a new copy and started again.  Changed the config file to be english and ran it - no problem.

Changed the config file to this (without comments)
``` 
sleep_begin_hour = 22
sleep_begin_minute = 00
sleep_end_hour = 07
sleep_end_minute = 00
sleep_brightness = 10
```

And it crashes - big red x  on the clock and in terminal I'm getting 

```
Error: In plugin time_default
Running plugin time_default
Error: In plugin time_default
Running plugin time_default
....
```

If I add a comment on each line of the variables again it runs fine.  Obviously without the brightness reduction.

phenze, 2019-12-03:
try printing the exact exception
Find the error handling code in wordclock.py and replace it with the following:

```python
except Exception as e:
    print(e)
    print('ERROR: In plugin ' + self.plugins[self.plugin_index].name + '.')
    self.wcd.setImage(os.path.join(self.pathToGeneralIcons, 'error.png'))
````

Then you will see a detailed error message. Try to fix it or paste it here then i will help you

SteveThomson19, 2019-12-03:
That helps greatly. Thank you.

The error comes up as :

`wordclock_display instance has no attribute 'getPixelColor`

If I look in wordclock_display the only reference to it is on line 121.

```
for i in range(self.wcl.LED_COUNT):
            color = self.**getPixelColor**(i)
            color = (color/brightness_before)^(1/2.2) * brightness
            self.setPixelColor(i, color)
```

I can't see where this would be referenced from originally - is it part of the rpi281x load ?

phenze, 2019-12-03:
I had this error too in my fork after pulling the changes.
I fixed this with the old code (wordclock_display.py)
```python
def setBrightness(self, brightness):
        """
        Sets the color for a pixel, while considering the brightness, set within the config file
        """
        brightness_before = self.getBrightness()
        brightness = max(min(255, brightness), 0)

        # for i in range(self.wcl.LED_COUNT):
        #     color = self.getPixelColor(i)
        #     color = (color/brightness_before)^(1/2.2) * brightness
        #     self.setPixelColor(i, color)

        self.strip.setBrightness(brightness)
        self.brightness = brightness
        self.show()
```
But this is something @bk1285 must have a look at.
This has to be fixed.

bk1285, 2019-12-03:
I'll look into it! Thanks for reporting.

bk1285, 2019-12-03:
Hi @SteveThomson19, @phenze,

can you checkout https://github.com/bk1285/rpi_wordclock/tree/bugfix/issue141 and check, whether this fixes the issue?

Best,
Bernd

phenze, 2019-12-04:
Uff that was on the develop branch and a lot of changes.

But it is almost fixed. When brightness before is 0 you get a division by zero exception.

My Fix (perhaps you dont need the developer mode switch)
```python
def setBrightness(self, brightness):
        """
        Sets the color for a pixel, while considering the brightness, set within the config file
        """
        brightness_before = self.getBrightness()
        brightness = max(min(255, brightness), 0)
        if self.developer_mode:
            if brightness_before > 0:
                for i in range(self.wcl.LED_COUNT):
                    neoPixelColor = self.strip.getPixelColor(i)
                    blue = ((neoPixelColor & 255)/brightness_before) * brightness
                    green = (((neoPixelColor >> 8) & 255)/brightness_before) * brightness
                    red = (((neoPixelColor >> 16) & 255)/brightness_before) * brightness
                
                    color = wcc.Color(red, green, blue)
                    self.strip.setPixelColor(i, color.neopixel())
            else:
                color = wcc.Color(0, 0, 0)
                for i in range(self.wcl.LED_COUNT):
                    self.strip.setPixelColor(i, color.neopixel())
        else:
            self.strip.setBrightness(brightness)
        self.brightness = brightness
        self.show()
```

But now i have another problems.
You forgot the set brightness call on the show_time function:
```python
def show_time(self, wcd, wci):
        now = datetime.datetime.now()
        # Set background color
        wcd.setColorToAll(self.bg_color, includeMinutes=True)
        # Returns indices, which represent the current time, when being illuminated
        taw_indices = wcd.taw.get_time(now, self.purist)
        wcd.setColorBy1DCoordinates(taw_indices, self.word_color)
        wcd.setMinutes(now, self.minute_color)
        wcd.setBrightness(self.brightness_mode_pos)
        wcd.show(self.typewriter and now.minute % 5 == 0)
```

With that fix every Minute my LED's are shown but then get invisible.
That is because of the new Show function in wordclock_display.py

```python
def show(self, animation = None):
        """
        This function provides the current color settings to the LEDs
        """
        fps = 10

        if animation is None or self.developer_mode:
            self.transition_cache_curr = self.transition_cache_next
            self.render_transition_step(self.transition_cache_curr)
        elif animation:
            transition_cache = wordclock_screen.wordclock_screen(self)
            for y in range(self.get_wca_height()):
                for x in range(self.get_wca_width()):
                    if self.transition_cache_next.matrix[x][y] is not wcc.BLACK:
                        transition_cache.matrix[x][y] = self.transition_cache_next.matrix[x][y]
                        self.render_transition_step(transition_cache)
                        sleep(0.12)
            self.transition_cache_curr = self.transition_cache_next
            self.render_transition_step(self.transition_cache_next)
        else:
            transition_cache = deepcopy(self.transition_cache_curr)

            brightness = self.getBrightness()
            while self.getBrightness() > 0:
                self.setBrightness(self.getBrightness() - 1)
                sleep(1/fps)

            self.transition_cache_curr = self.transition_cache_next
            self.render_transition_step(self.transition_cache_curr)
```

Can you please explain to me the else in that statement ? Why are you setting the brightness ? and never set back ?

SteveThomson19, 2019-12-04:
New cloned copy.
Only configuring the brightness parameters.
I get the Red X and the same error message I got thanks to Pascals suggestion.

`wordclock_display instance has no attribute 'getPixelColor'`

DJ-Dan, 2019-12-31:
I have the same problem with my new wordclock installation. The Sleep time won't be set and used. How can i fix it?

SteveThomson19, 2020-05-11:
Don't suppose we're any closer to closing this one are we ?

sevjordan, 2020-11-01:
Hi all, i am experiencing the same problem. Would be nice if this feature works for getting the clock to sleep at night =).
If there is any solution let us all know :-)

sezijlema, 2021-02-24:
A workaround for anyone with sleep issues (myself included) is using the API and crontab.

Run `sudo crontab -e` and add the following lines:

`0 23 * * * curl -X POST "http://localhost/api/plugin" -H "accept: application/json" -H "Content-Type: application/json" -d "{ \"name\": \"leds_off\"}"`

`0 7 * * * curl -X POST "http://localhost/api/plugin" -H "accept: application/json" -H "Content-Type: application/json" -d "{ \"name\": \"time_default\"}"`

Now the clock LEDs turns off at 23:00 and turn on at 7:00. You can edit the times if you'd like and even set different schedules for [weekdays](https://crontab.guru/weekdays-only) and [weekends](https://crontab.guru/weekends-only).

Alternatively, if you want to keep the clock on with a low brightness, you can also set the brightness:
`curl -X POST "http://localhost/api/brightness" -H "accept: application/json" -H "Content-Type: application/json" -d "{ \"brightness\": 10}"`

For more API options, visit http://YOUR_CLOCK_IP/api .

bk1285, 2021-08-09:
Hi all,

I recently refactored the implementation to send the clock to sleep. Should be fixed on latest develop (on master soon).

Best,
Bernd

closed 2021-08-09 by bk1285