bk1285/rpi_wordclock #145: Warning after start

DJ-Dan, 2019-12-19:
If I start the program on my Rpi it gives out an warning.
`WARNING: This is a development server. Do not use it in a production deployment.`
and the http-address shows http://0.0.0.0:80/
and i can't close the programm with Ctrl + C

oxivanisher, 2019-12-30:
This is probably OK. This message appears, because Flask (the web framework) is meant to run big websites with lots of connections. For such a environment, it is good practice to use other things together with Flask.
But since this is only the control web interface where 99% of the time nothing is happening, it is absolutely reasonable to use Flask like it is used here. This warning can be ignored without any problems.

DJ-Dan, 2019-12-31:
> This is probably OK. This message appears, because Flask (the web framework) is meant to run big websites with lots of connections. For such a environment, it is good practice to use other things [...]

Thanks @oxivanisher 
Can you also explain to me why I cannot close the application with CTRL + C?

oxivanisher, 2019-12-31:
@DJ-Dan Probably, but I can sure help you to close it. :D
It probably forkes (makes additional processes in the background) or is in a loop. When I wanted to close it, I just had to spam CRRL+C several times and I was able to "break out" of it. Hope this helps. :)

closed 2020-01-01 by bk1285