bk1285/rpi_wordclock #146: cannot start Matrix_Time plugin

webdisaster, 2019-12-29:
Hi!

Leider taucht das matrix time plugin nirgends auf. Auch mit den Tasten lässt es sich nicht auswählen. in der config ist es aber aktiviert

bk1285, 2020-01-01:
strange.... wie sieht's mit anderen Plugins aus? Die funktionieren? Gibt es irgendwelche Fehlermeldungen auf der Konsole?

Best,
Bernd

webdisaster, 2020-01-05:
Hi und Damke für die Hilfe. Die Fehlermeldung beim Start lautet:
Failed to import plugin time_matrix!

alle anderen Plugins gehen.
fg
Bernhard

Sofradex, 2020-09-23:
Hi there,
I also started to build one; total noob regarding to linux and python. Managed to get most of the plugins loaded; running on rasbian stretch 13-11-2018. When i run the program, the minute-lights seem to work properly, but the lights of the WCA seem to light up randomly (should be dutch), they change every 5 mintes, but there is nog recognisable pattern; could this be reated with the "Failed to import time_matrix"-error?

GPIO interface disabled. If hardware buttons are attached, any input is ignored. Webinterface can be used instead.
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to dutch.
Skipping plugin tetris since it is set to activate=false in the config-file.
Imported plugin 0: "leds_off".
Imported plugin 1: "restart".
Skipping plugin feed_parser since it is set to activate=false in the config-file.
Imported plugin 2: "time_as_words_dutch".
Imported plugin 3: "sunrise".
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be imported.
Imported plugin 4: "rainbow".
  INFO: No activate-flag set for plugin time_in_seconds within the config-file. Will be imported.
Imported plugin 5: "time_in_seconds".
Skipping plugin shutdown since it is set to activate=false in the config-file.
Failed to import plugin time_matrix!
Imported plugin 6: "ip_address".
Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
Imported plugin 7: "matrix".
  No sleeping time set, display will stay bright 24/7.
  No sleeping time set, display will stay bright 24/7.
  No sleep brightness set within the config-file. Defaulting to 5.
  Selected "time_default" as default plugin
Imported plugin 8: "time_default".
Skipping plugin weather_forecast since it is set to activate=false in the config-file.
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
Running plugin time_default.

FrankX0, 2020-09-23:
I doubt that your issue is related to the failure to load the plugin.
IMO it could be related to two things:
1) Incorrect addressing of the LED's
Can you make sure the routing of the LED's is per the instructions? So check for mirroring or rotation.
Do the minute LED's turn on in the correct order? Top-left, top-right, bottom-left, bottom-right.
2) Hardware (noise) issue
In this case one would expect random LED's to turn on, in random colours. And the minute-LED's would also be affected.

Sofradex, 2020-09-23:
Hm…

  1.  My order: bottom-right, top-left, bottom left, top-right. So that’s defentely not good. Should I choose another wiring layout? Or rebuild the minute leds?
  2.  Seems ok. All lights are white, no random colours

Thank you for your help!

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

Sofradex, 2020-09-23:
AAAH.

Figgured it out allready… My clock measures 10X11 instead of 10X11….

Thanks very much

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

FrankX0, 2020-09-23:
Opgelost dus?

Sofradex, 2020-09-23:
Ja, tenminste dat denk ik.

Ga de ledjes helemaal opnieuw leggen. Dat was in ieder geval niet goed. Weet niet of daarmee het “failed to import time_matrix” probleem is opgelost, maar dat zien we dan wel. Heb er goede hoop op.

Groeten, Hein

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

FrankX0, 2020-09-23:
Hoi Hein,

Het “failed to import time_matrix” zal daarmee niet opgelost zijn.
Zal wel eens kijken of ik het probleem kan vinden.

Mvg,
Frank.

Sofradex, 2020-09-24:
Ik heb bij het installeren de installer.sh gebruikt; misschien had ik dat niet moeten doen?

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

FrankX0, 2020-09-25:
@webdisaster @Sofradex 
I just created a pull request to fix this issue.
If needed you can also make this change locally (replacing a TAB with spaces)..

Sofradex, 2020-09-30:
Got it all working now, special thanks to Frank!

FrankX0, 2020-09-30:
Good to hear!
@bk1285 can you merge pull request #168 so we can close this issue?

bk1285, 2021-01-15:
Nice, @FrankX0!

closed 2021-01-15 by bk1285