bk1285/rpi_wordclock #147: html-string

DJ-Dan, 2020-01-01:
Is it possible to change the LED's on and off with html string?
I want control the wordclock with my FHEM System.

FrankX0, 2020-01-01:
I guess what you are looking for is the build-in API.
Please have a look at the API documentation in web interface of the clock: <ip-address>/api

bk1285, 2020-01-01:
Hi @DJ-Dan 

if you manage to control the clock via FHEM, it would be great, if you'd write about your solution here. Ideally, you can use the endpoints as outlined by @FrankX0 e.g. using `curl` or similar tools.

I did the integration for home assistant. Maybe you want to have a look [here](https://github.com/bk1285/rpi_wordclock_for_homeassistant/blob/master/rpi_wordclock.py) to get at least an idea, which endpoints you can use for your integration.

Best,
Bernd

DJ-Dan, 2020-02-01:
> I guess what you are looking for is the build-in API.
> Please have a look at the API documentation in web interface of the clock: /api

Hey @FrankX0, I don't know what you mean?
Can you describe me how i integrate the API in my FHEM or FHEM into the Wordclock?

/br
Daniel

FrankX0, 2020-02-02:
Hi Daniel,

If you want to control the clock from FHEM, you can send html POST requests to the IP address of the clock. The commands support are documented in the clock webinterface.
You can find it at: `your-clock-ip-address/api`

I however have no experience with FHEM. So I cannot help you with how to send POST requests from FHEM.
Maybe this helps: https://wiki.fhem.de/wiki/HTTPMOD

Good luck,
Frank.

DJ-Dan, 2020-02-03:
Thanks for your reply.
If I edit the brightness value over the API website, then I become a error message:
error code 500
"message": "Internal Server Error"
Does anyone know how i fix this?

FrankX0, 2020-02-03:
How and which POST request do you send?
Can you share the output of the terminal window (if you start the clock application manually)?

bk1285, 2020-02-03:
> How and which POST request do you send?
> Can you share the output of the terminal window (if you start the clock application manually)?

... that would be indeed helpful. A "Internal Server Error", indicates definitely a wordclock issue. Even if you submit any wrong values, etc, the wordclock should be able to handle that.

Best,
Bernd

DJ-Dan, 2020-02-07:
Hi,
this is the output when i start the clock manually,

`Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: christians_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to german2.
Imported plugin 0: "sunrise".
  No sleeping time set, display will stay bright 24/7.
  No sleeping time set, display will stay bright 24/7.
  No sleep brightness set within the config-file. Defaulting to 5.
  Selected "time_default" as default plugin
Imported plugin 1: "time_default".
Failed to import plugin time_matrix!
  INFO: No activate-flag set for plugin time_in_seconds within the config-file. Will be imported.
Imported plugin 2: "time_in_seconds".
Imported plugin 3: "feed_parser".
Imported plugin 4: "restart".
Imported plugin 5: "leds_off".
Imported plugin 6: "time_as_words_dutch".
Imported plugin 7: "matrix".
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be imported.
Imported plugin 8: "rainbow".
  Assumes no temperature sensor to be attached.
Imported plugin 9: "weather_forecast".
Imported plugin 10: "ip_address".
Imported plugin 11: "shutdown".
Imported plugin 12: "time_as_words_german".
Imported plugin 13: "tetris".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
Running plugin time_default.
`
and...
this is the output when I change the brightness to the value 10 over the api post side. 

`10.175.0.47 - - [07/Feb/2020 09:50:10] "GET /api HTTP/1.1" 200 -
10.175.0.47 - - [07/Feb/2020 09:50:10] "GET /swaggerui/droid-sans.css HTTP/1.1" 304 -
10.175.0.47 - - [07/Feb/2020 09:50:10] "GET /swaggerui/swagger-ui-standalone-preset.js HTTP/1.1" 304 -
10.175.0.47 - - [07/Feb/2020 09:50:10] "GET /swaggerui/swagger-ui-bundle.js HTTP/1.1" 304 -
10.175.0.47 - - [07/Feb/2020 09:50:10] "GET /swaggerui/swagger-ui.css HTTP/1.1" 304 -
10.175.0.47 - - [07/Feb/2020 09:50:11] "GET /api/swagger.json HTTP/1.1" 200 -
[2020-02-07 09:50:28,921] ERROR in app: Exception on /api/brightness [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/api.py", line 325, in wrapper
    resp = resource(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/flask/views.py", line 89, in view
    return self.dispatch_request(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/resource.py", line 44, in dispatch_request
    resp = meth(*args, **kwargs)
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 199, in post
    web_interface.app.wclk.wcd.setBrightness(brightness)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 119, in setBrightness
    color = self.getPixelColor(i)
AttributeError: wordclock_display instance has no attribute 'getPixelColor'
10.175.0.47 - - [07/Feb/2020 09:50:28] "POST /api/brightness HTTP/1.1" 500 -
`

FrankX0, 2020-02-07:
Hi @bk1285, Bernd,

I guess this one is for you to solve: `wordclock_display instance has no attribute 'getPixelColor'`
And there is another occurrence of the same issue: #141 

Kind regards,
Frank.

bk1285, 2020-02-07:
Agreed, @FrankX0 ;) 

@DJ-Dan, can you provide me the commit, you're running the wordclock with? ...are you on master?

DJ-Dan, 2020-02-07:
> Agreed, @FrankX0 ;)
> 
> @DJ-Dan, can you provide me the commit, you're running the wordclock with? ...are you on master?

I'm not sure what you mean.
Do you want to have my wordclock config file?
Yes I'm the master of the system an i have full access to it.

bk1285, 2020-02-07:
No worries, @DJ-Dan,

if you login to your raspberry pi and open a terminal, can you provide me with the output of the following commands?
`cd /home/pi/rpi_wordclock`
`git rev-parse --short HEAD`

The last command should return a short sequence of characters. These are of interest here, since they identify your current software version (=current commit, you're on...)

Best,
Bernd

DJ-Dan, 2020-02-07:
> No worries, @DJ-Dan,
>
> if you login to your raspberry pi and open a terminal, can you provide me with the output of the following commands?
> `cd /home/pi/rpi_wordclock`
> `git rev-parse --short HEAD` [...]

The output is: 'd40506c'

bk1285, 2020-02-07:
Thanks for the quick feedback, @DJ-Dan. So I have all input to look into it.

Best,
Bernd

DJ-Dan, 2020-02-07:
> Thanks for the quick feedback, @DJ-Dan. So I have all input to look into it.
> 
> Best,
> Bernd

DJ-Dan, 2020-02-07:
upps i clicked myself!
Thanks for your support

bk1285, 2020-02-08:
Hey @DJ-Dan 

I made quite a few changes and refactorings, which should probably be tested a little, before going to master.
To check, if the fix is working, try the following commands:
` cd /home/pi/rpi_wordclock`
`git remote update`
`git checkout --track origin/develop`

You're now on the develop branch, which holds my latest fixes. If it doesn't work and you want to go back to the version you have now, run
`git checkout master`

Would be good to hear, whether your issue is resolved... ;)

Best,
Bernd

DJ-Dan, 2020-02-08:
Hey @bk1285 
the command `git remote update` worked fine but the `git checkout --track origin/develop` gave a error because i changed the wordclock_display.py fiel

I have generated a new code `ce689c4`

sorry for the inconvenience

bk1285, 2020-02-08:
Run
`git reset --hard` 
and then
 `git  checkout --track origin/develop`.
Note: The changes you made to wordclock_display.py are lost in this case.

DJ-Dan, 2020-02-08:
I hope I did everything right
The outputs from the codes are:

`pi@raspberrypi:~ $ cd rpi_wordclock/
pi@raspberrypi:~/rpi_wordclock $ git remote update
Fetching origin
pi@raspberrypi:~/rpi_wordclock $ git checkout --track origin/develop
fatal: A branch named 'develop' already exists.
pi@raspberrypi:~/rpi_wordclock $
pi@raspberrypi:~/rpi_wordclock $ git checkout master
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
pi@raspberrypi:~/rpi_wordclock $
pi@raspberrypi:~/rpi_wordclock $ git reset --hard
HEAD is now at 42d795d Updated installer script
pi@raspberrypi:~/rpi_wordclock $ git checkout --track origin/develop
fatal: A branch named 'develop' already exists.`

And the output from the api command is the same

bk1285, 2020-02-08:
:see_no_evil: ... so you already have a develop-branch from former tests/checkouts...? ;)

In this case, run
`git remote update`
`git checkout develop`
`git pull origin develop`

Just as remark: The current version of develop works best if `typewriter` is set to  `False` within the wordclock_config.cfg.

DJ-Dan, 2020-02-08:
Okay sorry for my fault.

`pi@raspberrypi:~ $ cd rpi_wordclock/
pi@raspberrypi:~/rpi_wordclock $ git remote update
Fetching origin
pi@raspberrypi:~/rpi_wordclock $ git checkout develop
Switched to branch 'develop'
Your branch is up-to-date with 'origin/develop'.
pi@raspberrypi:~/rpi_wordclock $ git pull origin develop
From https://github.com/bk1285/rpi_wordclock
 * branch            develop    -> FETCH_HEAD
Already up-to-date.
`

and yes it works. I can now set the brightness per api.
When i set a value the clock dim to zero and then to the value i set, but this is not at the same time i execute it. It is a delay, but thats not a problem, only for note!