bk1285/rpi_wordclock #151: Problems with Raspberry Pi Zero W and the W2812B LEDstrip

BowenDragonheart, 2020-01-06:
Hi WordClock makers!,

I am trying to build my own Word Clock and am in the final stage of testing the LED strips. Unfortunately, I ran into problems here.

I'm using a Raspberry Pi Zero W, combined with a Mean Well power supply (5V, 50W) and the a W2812B LEDstrip.

Using both the strandtest.py and the test script of the source directory, I run into a strange problem: I need to specifiy four more LEDs than are actually in the LEDstring to have all light up.

To investigate more, I made a simple program based on the strandtest.py and an issue raised in the rpi_ws281x github (https://github.com/jgarff/rpi_ws281x/issues/314) .

----------------------------------------------------------------------------------------------------

from neopixel import *

LED strip configuration:
LED_COUNT = 10 # Number of LED pixels.
LED_PIN = 18 # GPIO pin connected to the pixels (18 uses PWM!).
#LED_PIN = 10 # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000 # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10 # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 255 # Set to 0 for darkest and 255 for brightest
LED_INVERT = False # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0 # set to '1' for GPIOs 13, 19, 41, 45 or 53

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
strip.begin()

strip.setPixelColorRGB(0, 127, 127, 127)

strip.show()

----------------------------------------------------------------------------------------------------

If I run this, the first LED become white. So far so good. If also the second LED is activated,

strip.setPixelColorRGB(0, 127, 127, 127)
strip.setPixelColorRGB(1, 127, 127, 127)

then the first remains white, but the second becomes yellowish green. If also the third is given a signal

strip.setPixelColorRGB(0, 127, 127, 127)
strip.setPixelColorRGB(1, 127, 127, 127)
strip.setPixelColorRGB(2, 127, 127, 127)

Then the first and second become white and the third bright green.

To conclude

strip.setPixelColorRGB(0, 127, 127, 127)
strip.setPixelColorRGB(1, 127, 127, 127)
strip.setPixelColorRGB(2, 127, 127, 127)
strip.setPixelColorRGB(3, 127, 127, 127)
strip.setPixelColorRGB(4, 127, 127, 127)
strip.setPixelColorRGB(5, 127, 127, 127)
strip.setPixelColorRGB(6, 127, 127, 127)
strip.setPixelColorRGB(7, 127, 127, 127)
strip.setPixelColorRGB(8, 127, 127, 127)
strip.setPixelColorRGB(9, 127, 127, 127)

Results in that the first seven are white, the 8th is yellow and LED9 and LED10 are not giving light at all.

I tried with several LED strips of 10 and all gave the same result.

Who can help me to solve this? I also made an issue at the rpi_ws281x github, but there people appear not to be that active. 
.
You will have my eternal gratitude if you can help me with this.

Best regards,
Bart

plotaBot, 2020-01-07:
Hello @BowenDragonheart,

welcome to the Wordclock builders.
I have three things you might consider to get rid of that behaviour:
1. Are you using this commit of rpi_ws281x as described in [Software Setup](https://rpi-wordclock.readthedocs.io/en/master/doc_software_setup.html#rd-party-dependencies-git-repositories)?
[image]
2. Remove the level converter. (I am also using a Rpi Zero W and the LEDs light up fine witout one.)
3. Swap the Meanwell with an USB wall adapter. (Maybe this is a better and safer solution.)

Greetings, plotaBot

BowenDragonheart, 2020-01-07:
Thank you for your input!

1. Yes, but I will do a clean install right now just to make sure that everything.
2. I don't use a level converter.
3. I will give this a try.

Furthermore, I have read that it is important that the LED strip and the PI do not share the 5V+
(https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/)

This is currently the case for my clock. Why is this important? Can this explain the wrong behaviour?

plotaBot, 2020-01-07:
Ok. Have you connected both Ground (Power Pi AND Power LEDs) to the Pi?
I am not experiencing any kind of problems using one Powersupply. My wiring looks like this:
[image]

Maybe another thing causing this problem could be related to the beefy powersupply iterfering with the signals. btw: 10-15W should be enough and an ordinary phone charger is capable to deliver such power. Since you are not running all 114 LEDs at the same time on white with maximum brightness you should be fine. (I once measured this value and Pi + all LEDs were using <2A peak)

Let us know if anything I suggested works or not ;)

BowenDragonheart, 2020-01-07:
Wait, do you need also to connect the last LED to the +5V and the GND? :O

BowenDragonheart, 2020-01-07:
Is that not the problem? It is not an RGB strip, but a RGBW strip. Is this not supported?

plotaBot, 2020-01-07:
Apparently rpi_ws281x supports RGBW but implementing it requires more effort and a strip at hand. Sorry if you bought the wrong LED strip.
Unfortunately I am currently not able to develop a rpi_wordclock RGBW support (time and no LEDs). Or you, Bowen, are passionate enough to dig into that?

@bk1285: Would you add the enhancment label so @BowenDragonheart is able to use his LEDs (somewhere in the future).

bk1285, 2020-01-25:
Hi @BowenDragonheart 

to address the LEDs, jgarff's library is currently used. As mentioned [here](https://github.com/jgarff/rpi_ws281x/issues/387#issuecomment-574727119), it's probably an issue, that you need to provide another byte to setup the fourth color channel. As soon, as jgarff's library supports this, I expect the wordclock to work smoothly with th RGBW-type of LEDs.

Meanwhile, you'll have to choose standard RGB-strip.

Best,
Bernd

closed 2020-01-25 by bk1285