bk1285/rpi_wordclock #153: PCB

dhoeben, 2020-02-16:
Hi everyone,

I have a question. I'm interested in making this project for a long time, but finally found the time to do it now. I have no clue of whatsoever of Raspberry Pi, but a little of making printer circuit boards. Now my question regarding the parts, does it matter which LED's, level-shifter, etc I will use on the PCB?

I was thinking of these LED's:
https://lcsc.com/product-detail/Light-Emitting-Diodes-LED_5050-RGBIntegrated-Light-4Pin_C114586.html

and this level shifter:
https://lcsc.com/product-detail/Level-Translators-Shifters_Texas-Instruments_TXS0108ERGYR_Texas-Instruments-Texas-Instruments-TXS0108ERGYR_C90706.html

I'd love to hear your thoughts!

If some of you are interested, I will keep my progress going here! I think it might even be (a little) cheaper and more reliable?

Cheers!

FrankX0, 2020-02-17:
Regarding the LED's: are you planning to use separate LED's instead of the strips? That would mean a PCB the size of the clock? Neat, but maybe much more expensive.

Regarding the level shifter: you only have to go from a 3.3V to a 5V domain. I would advise to use a 5V "buffer" with TTL compatible inputs. So something like a (less exotic) [HCT08](https://lcsc.com/product-detail/Gates-and-Inverters_Texas-Instruments_SN74HCT08DR_Texas-Instruments-Texas-Instruments-SN74HCT08DR_C6767.html).

Good luck with your project!

dhoeben, 2020-02-18:
Yes, I was planning on making the PCB almost as big as the clock and cut a piece out of it for Raspberry! I was looking, here in the Netherlands it is around 10-15 euro for a printed one with parts. Of course, depending on the LED’s etc used.

Thanks for the input! Will keep you posted!

NemoN, 2020-02-18:
> Yes, I was planning on making the PCB almost as big as the clock and cut a piece out of it for Raspberry! I was looking, here in the Netherlands it is around 10-15 euro for a printed one with parts. Of course, depending on the LED’s etc used.

I am interested :-)

dhoeben, 2020-02-18:
Just out of curiosity, why would you use a level shifter? Is this because of how the Pi handles the led communication? [Here](https://rpi-wordclock.readthedocs.io/en/master/_images/wiring.png) they don't list it...

FrankX0, 2020-02-18:
Here's the explanation why: https://github.com/bk1285/rpi_wordclock/issues/107#issuecomment-447108988
If you're lucky, it will work without. But by design it is not reliable. If you are designing a custom PCB anyway, stay on the safe side.

> cut a piece out of it for Raspberry

I would advise to go for a Raspberry Pi Zero W and connect it through the IO connector to your PCB ("piggy-back"). The Pi then can be powered directly from your PCB through the IO connector (no micro-USB connection required).

dhoeben, 2020-02-18:
@FrankX0 Oke thanks for the clarification about the level shifter.

> I would advise to go for a Raspberry Pi Zero W .........
Is the Pi Zero W also protected for overvoltage etc like via the USB connection?

FrankX0, 2020-02-18:
The Pi Zero has no overvoltage protection. The Pi4 has a TVS on its 5V supply.
Then again, you can also add protection on your PCB.

bk1285, 2021-07-24:
Hey @dhoeben,

any updates regarding the PCB? ...sound's pretty promising. Feel free to submit pictures at any time... ;)

Best,
Bernd

kitesurfercharlie, 2022-03-31:
I had a go at this and forgot to set Power flags in KiCad 4. As a result the autorouter forgot to connect all the ground and power connections. At some point I will have another go.  Here is the pcb... https://youtu.be/BwkPAvACl9Y