bk1285/rpi_wordclock #154: initialization of new brightness handling

bhunziker, 2020-02-23:
I get an exception with the new brightness handling while initializing class `wordclock_display`.
Assuming that the exception comes from `setBrightness ` in initialization (line 36) when it will call `getBrightness `(line 115) but self.brightness (line 109) is not available at this time.
My proposal: `self.brightness = config.getint('wordclock_display', 'brightness')` on line 36.

bk1285, 2020-02-27:
Hi @bhunziker,

are you on latest master?
Can you post the result of the commands below?

```
cd /home/pi/rpi_wordclock
git rev-parse --short HEAD
```

Best,
Bernd

Arjan444, 2020-02-28:
Hi Bernd,

I'm on 42d795d.

I also have a brightness issue. When starting the clock I get the message that brightness is not set ("Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.")

But the brightness setting in set. It is untouched in the config (and set at 200). 

I also see that the sleeping time is not set (but it is). 
Can it be that the config file is not properly?

Thanks again. Arjan

bhunziker, 2020-03-05:
Hi Bernd

Sorry for the delay in responding. I was abroad and had no access to my wordclock.
I'm on 42d795d. The description of Arjan444 holds for me as well. I assume it's the same root cause, the initialization routine.

Thanks
Bernhard

bk1285, 2020-04-25:
Hi @bhunziker, 

same as #155 : I'll take a look at this and provide a fix on latest develop. On develop, I'm currently refactoring the wordclock's code quite a bit, including the brightness handling.

Cheers,
Bernd

closed 2020-04-25 by bk1285