bk1285/rpi_wordclock #155: Startup Brightness and color

Azrael0815, 2020-02-29:
First of all I would like to say thank you for the great job.. i created your wordclock and it looks awesome !

Nevertheless I have a (probably stupid) question. How do I change the startup brightness and color ?
The "brightness" in the cfg file does not seem to affect the brightness at all
Thanks for your time andf great work

TobyLongley, 2020-03-16:
Hello. 
I have the same problem. I finished my wordclock this weekend and tried to adjust the brightness as well in the wordclock_config.cfg, time_default/plugin.py and  wordclock_display.py but nothing had any influence. 
The only way to change the brightness is with the webapp atm.

Any hints where to start? 

Thank you in advance.

Azrael0815, 2020-03-18:
I managed this by adding a brightness sensor like Phenze did. Works really good. Still I am only able to change the startup color by changing the code in the Py file

TobyLongley, 2020-03-21:
> 
> 
> I managed this by adding a brightness sensor like Phenze did. Works really good. Still I am only able to change the startup color by changing the code in the Py file

noobquestion: where do I find his code? I'm pretty new to all this git stuff :/

spintike, 2020-03-30:
Hi @TobyLongley.

You have to add the following line in your wordclock_config.cfg file:

https://github.com/phenze/rpi_wordclock/blob/d52ce76d5633a182cf8192a1f75848dbee45aff2/wordclock_config/wordclock_config_old.cfg#L33

Then the wordclock script execution should show something like the following when the rest of the requirements are met:

`Using brigtness sensor : True
Importing sensor Library`

TobyLongley, 2020-04-13:
Hello guys,

sorry for the late reply I had no time to look through the code...

Thank you really much for the help but it still didn't work :/
I'm using another brightnesssensor therefore phenzes code doesn't work for me. I do manage to get a sensible value from my sensor but I am still struggling to get it into the wordclock code -.-

And I'm a python newb so maybe someone can explain to me how this should normally work. What i think:

- `time_default/plugin.py` gets loaded
- the variable for the brightness is `brightness_mode_pos`
- in [line 222 ](https://github.com/bk1285/rpi_wordclock/blob/42d795dcb6058ba6420205053f1dc8fa4969e8a3/wordclock_plugins/time_default/plugin.py#L222)  `wcd.setBrightness` is called
- in `wordclock_tools/wordclock_display` the `setBrightness` method is called, there is this small iteration `for i in range...` and in line 119 `self.getPixelColor` 

but there is no method called getPixelColor defined at all? how is that supposed to work?

I thought that this might be a bug so I added

```
    def getPixelColor(self, pixel):
        return self.strip.getPixelColor(pixel)
```

Now it kind of works but instead of changing the brightness of the LEDs, the wordclock changes the color of the LEDs (like from whitewarm to blue, to another blue etc...but I think thats a problem I have to fix by myself

phenze, 2020-04-14:
Just switch to my master.
I have fixed that. That is a Bug on the current Master of Bernd.

When you want to fix that on your own do the following:
1. Comment out the loop
2. Add the old set Brightness call (i have marked the line for you in the screenshot)
[image]

TobyLongley, 2020-04-14:
> Just switch to my master.
> I have fixed that. That is a Bug on the current Master of Bernd.
> 
> When you want to fix that on your own do the following:
> 
> 1. Comment out the loop
> 2. Add the old set Brightness call (i have marked the line for you in the screenshot)
>    [image]

Thank you! I will try that out in the evening.

bk1285, 2020-04-25:
I'll take a look at this and provide a fix on latest [develop](https://github.com/bk1285/rpi_wordclock/tree/develop). On develop, I'm currently refactoring the wordclock's code quite a bit, including the brighness handling.

Cheers,
Bernd

bk1285, 2022-09-01:
Should be resolved by now. If not, please reopen.

Best,
Bernd

closed 2022-09-01 by bk1285