bk1285/rpi_wordclock #156: WordClock wont startup

Alphascrypt, 2020-03-18:
Hey guys, i started to build this nice wordclock. I have all things here. I wanted to start the wordclock.py but it ended in an failure message:

 sudo python wordclock.py
Traceback (most recent call last):
  File "wordclock.py", line 10, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 3, in <module>
    from flask_restplus import Api, Resource, fields
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/__init__.py", line 4, in <module>
    from . import fields, reqparse, apidoc, inputs, cors
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/fields.py", line 17, in <module>
    from werkzeug import cached_property
ImportError: cannot import name cached_property

Its the new Raspbian Buster Lite version. 

Did i must use an older version of raspbian?

Greetings
Martin

phenze, 2020-03-18:
https://github.com/noirbizarre/flask-restplus/issues/777

You can either wait for flask_restplus to get fixed or install flask_restx

Azrael0815, 2020-03-18:
Or you switch to a different "Werkzeug" (0.16.1) version - worked for me
https://github.com/noirbizarre/flask-restplus/issues/777

Alphascrypt, 2020-03-23:
#Worked! Thank you

bk1285, 2020-04-25:
Thank you for the solution, @phenze 

... so no action required here, since the issue will be fixed by updated external dependencies.

closed 2020-04-25 by bk1285