bk1285/rpi_wordclock #157: DCF77

Alphascrypt, 2020-03-24:
Hey guys again,
is it possible to implement an DCF77 for the time without WLAN?

Greetings
Martin

FrankX0, 2020-03-26:
The wordclock uses the internal time of the Raspberry Pi. Normally it is updated and synchronised with an NTP server over (W)LAN. In your case this source should be through DCF77.
So as far as I can see the wordclock is not directly affected by how the Raspberry Pi time is being kept accurate.

So then your question is how to synchronise the Raspberry Pi time through a DCF77 module?

Alphascrypt, 2020-03-26:
Hey @FrankX0 ,
the raspberry system time should be syncronised. I would like to use the wordclock without wifi and with wifi. 

Auf deutsch isses leichter :P also es sollen beide zeiten genommen werden, einmal vom dcf und vom ntp. Wenn kein wlan verfügbar ist soll die zeit vom dcf genommen werden.

Gruß
Martin

FrankX0, 2020-03-26:
Interesting.
Did you already do some research how to implement a DCF77 receiver (maybe from [Conrad](https://www.conrad.de/de/p/dcf-empfaenger-modul-641138-passend-fuer-serie-c-control-641138.html))?
And why would you switch back to WLAN if DCF77 is always available?
Normally I expect the Raspberry Pi clock to be accurate within a couple of seconds per day. So synchronising every few days would be sufficient.

Gruß,
Frank.

Alphascrypt, 2020-03-26:
The DCF77 is not 100% accurate. Per day it can be 10minutes forward or backward. 
Implementing with only the dcf works but when its on wifi the dcf77 is the main clock, so the clock is not 100% accurate. 

I have the dcf77 from pollin. 

Syncronising without wifi is hard :P it would be nice to have an option with or without dcf77. Is this possible?

Maybe when no wifi is available make a Hotspot as well? For configuration etc?

Gruß
Martin

FrankX0, 2020-03-26:
Why do you think DCF77 is not accurate?
According Wikipedia, which is in line with my understanding of the (slightly) outdated time distribution system:

> The DCF77 transmitted carrier frequency relative uncertainty is 2 × 10−12 over a 24-hour period and 2 × 10−13 over 100 days, with a deviation in phase with respect to UTC that never exceeds 5.5 ± 0.3 microseconds.

I searched around and found some interesting sources:
[NTP Server via DCF77 on a Raspberry Pi](https://weberblog.net/ntp-server-via-dcf77-on-a-raspberry-pi/): this implements a local NTP Server by using DCF77 (because of its accuracy!)
[set-rtc-using-dcf77-via-dcf1](https://github.com/demogorgi/set-rtc-using-dcf77-via-dcf1/): a Python implementation on a ESP8266 using the receiver you have (I think).

My suggestion would be to first be able to have your clock synchronised with your DCF77 receiver.
And then (if still necessary) start thinking about your other wishes: switching between WLAN and DCF77 and a hotspot.

Alphascrypt, 2020-03-26:
Nu weiter auf deutsch. 

Die genauigkeit des DCF77 ist hier nicht die problematik. Die dekodierung des Raspberry sowie die daraus entstehenden timeouts. 

Die uhr läuft derzeit mit dem DCF77 und synchronisiert sich auch. Das geht alles ohne größere problematiken. 

Vielleicht von mir doof erklärt, Der DCF77 könnte als option hinzugefügt werden. Sowie die option mit WLAN oder DCF77 oder beides.

bk1285, 2020-12-19:
Hi @Alphascrypt

thank you for the update. Sounds interesting to not necessarily depend on wifi/internet to run the clock. Can you provide a link to the hardware you used -- maybe combined with a picture and possible does and don'ts? I'd consider adding that to the docs.

Best,
Bernd