bk1285/rpi_wordclock #159: [Feature request] Parameterize matrix dimension

a-sp, 2020-05-01:
Hello @bk1285m thank you very much for sharing your project! I really like it!

I have a little request... Would it be possible to parameterize the matrix dimension? 

Using Patrick's [WorldClockGenerator ](https://github.com/im-pro-at/WordClockGenerator) seems like that for Italian a 13x11 matrix is needed (the current layout is ok and totally understandable, but it is not how we tell time in Italy... we use looong words :D ). 

Thanks :)

plotaBot, 2020-05-02:
Hello @a-sp,
it is a pleasure to see the wordclock language support growing by users interested in adding new ones!
Please have a look at [this](https://github.com/bk1285/rpi_wordclock/issues/138#issuecomment-553535343) where Bernd described how you are able to add your own languge. (if you aren't already familiar with the procedure). He also mentioned a link to a helpful website from the creators of the original clock. It seems like they are able to use the original 11x10 matrix to display a readable italian time... look at [this](https://files.qlocktwo.com/Q2/check.html?color=BLACK%20ICE%20TEA&controls=false&language=IT&size=450) ;)
I don't know if this able to fullfills your needs.

a-sp, 2020-05-02:
Hi @plotaBot, thanks a lot  for your suggestions!

I already saw the original Qlocktwo Italian layout (and I made the time_italian.py myself before seeing @pueppixxxx pull request -.-' ). 
This layout is ok and readable, but it is not exactly how we tell time in Italian.

Main issues: 
-  Italian "o' clock" is missing;
- for 03:35 we don't say "25 to 4", we say "35 past 3", but with the current layout there is not enough space to write "35" in letters, 12 characters are needed ;
- we use special words for 12:00 PM/AM (nice to have).

My fear is that making the matrix dimension configurable can be a pain in the ass for additional plugins... the dirty solution might be to make matrix dimension configurable only for the time function, while all the other plugin would scale down to the original 11x10 size.

Joeri-MS, 2020-10-10:
Hope this helps.

There are 2 small changes in pi_wordclock/wordclock_plugins/matrix/plugin.py:
Line 38: 
`        rain = [20 for _ in range(0, wcd.get_wca_width() + 1)]`

Line 51:
`                    y1 = min(wcd.get_wca_height() + 1, y);`

Also in the file from the time_matrix/plgin.py

bk1285, 2020-12-19:
Hi @a-sp 

I was quite busy in 2020. ... hope that changes in 2021.

Were you able to resolve your issue -- e.g. considering @Joeri-MS comment? The software should be ready to handle varying matrix resolutions.

... this was however (to my knowledge) never tested. Therefore you might run into issues, when trying the first time. Feel free to report back here.

Best,
Bernd

a-sp, 2020-12-20:
Dear @Joeri-MS and @bk1285 thank you very much for getting back to me! It's really to receive a reply even after some time  ☺
Unfortunately, I've had to put the project on hold due to lack of free time, but I still would like to complete it... let's hope that 2021 will give us the opportunity to have more time to spend on projects we love! 

I'll keep you updated 😎

Joeri-MS, 2021-01-23:
The last problem i have is that the collor of the is not the same.
This only happens when the time or text (I think the function setColorBy2DCoordinates() is it) is show.
Images (Sunrise....) are fine, so i don't thin it is my hardware setup...

The word FÜNF is like warm white, the other more yellow:
[image]
[image]

Perhaps i will find it some day.

phenze, 2021-01-25:
Always check the following when such errors occur:
* Do you have common grounding ?
* Do you use a level shifter ? I think this is neccessary with that much LED's
* Try to connect +/- at multiple points at the strip!
* What power supply are you using? With your 18x16 Matrix you need 0,06A * (16*18 LED) =  17,28A
* The LED's need the most power when they are white. So when your clock is running fine when it is blue for example your problem is most likely the power supply

Joeri-MS, 2021-01-25:
It seems to bee the grounding, now it looks good, thanks.