bk1285/rpi_wordclock #160: Cannot Import name cached_property

MaxZimmermann99, 2020-06-15:
Hey guys,

trying to set up my wordclock i get the following error when starting "wordclock.py":

[image]

I already tried to downgrade Werkzeug to 0.16.1 (saw this somewhere online) and installing pyton 3.6. 

Someone with same problems out there? Any ideas?

Greetings

Max

schweizschweiz, 2020-07-30:
i do have the same problem. no solution yet

phenze, 2020-07-31:
What is your output of:
`pip show werkzeug`

Python3 will not work with the wordclock software. So be sure `python --version` will show something with 2.7

schweizschweiz, 2020-07-31:
output of `python --version` is 2.7.13 so it should be fine.
`pip show werkzeug` outputs 

Name: Werkzeug
Version: 1.0.1
Summary: The comprehensive WSGI web application library.
Home-page: https://palletsprojects.com/p/werkzeug/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD-3-Clause
Location: /home/pi/.local/lib/python2.7/site-packages
Requires:
`

phenze, 2020-08-03:
Why this didn't work ?
What is the error when trying to Install Werkzeug with the following command:
`pip install 'Werkzeug==0.16.1'`

schweizschweiz, 2020-08-03:
i reinstalled werkzeug 0.16.1 and verified with `pip show werkzeug`
Name: Werkzeug
Version: 0.16.1
Summary: The comprehensive WSGI web application library.
Home-page: https://palletsprojects.com/p/werkzeug/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD-3-Clause
Location: /home/pi/.local/lib/python2.7/site-packages
Requires:

the error for the wordclock programm ist still the same:
Traceback (most recent call last):
  File "wordclock.py", line 10, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 3, in <module>
    from flask_restplus import Api, Resource, fields
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/__init__.py", line 4, in <module>
    from . import fields, reqparse, apidoc, inputs, cors
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/fields.py", line 17, in <module>
    from werkzeug import cached_property
ImportError: cannot import name cached_property

phenze, 2020-08-03:
You perhaps have to install another flask-restplus version
Try:
`pip install 'flask-restplus==0.12.1'`
and
`pip install 'Flask==1.0.2'`

schweizschweiz, 2020-08-03:
i just uninstalled flask and flask-restplus and installed the version you've mentioned. still not working:(
i have flask version 1.0.2 and flask-restplus 0.12.1. again, verified with `pip show Flask` and `pip show flask-restplus`.

but using `sudo python rpi_wordclock/wordclock.py` outputs the same error again. 
Traceback (most recent call last):
  File "rpi_wordclock/wordclock.py", line 10, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 3, i                n <module>
    from flask_restplus import Api, Resource, fields
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/__init__.py", line                 4, in <module>
    from . import fields, reqparse, apidoc, inputs, cors
  File "/usr/local/lib/python2.7/dist-packages/flask_restplus/fields.py", line 1                7, in <module>
    from werkzeug import cached_property
ImportError: cannot import name cached_property

is there a way to uncomment the webinterface? as far as i understand that's the root of the problem

phenze, 2020-08-03:
I think you have a python installation problem.
The output of `pip install 'Werkzeug==0.16.1'` was the following path:
`Location: /home/pi/.local/lib/python2.7/site-packages`

But when you run the wordclock software is is using this path:
`File "/usr/local/lib/python2.7/dist-packages/flask_restplus/fields.py"`

So maybe this is the problem. Pip installs the packages to the wrong location and so the incorrect version is used.

schweizschweiz, 2020-08-03:
i wiped the pi zwero w and reinstalled everything. the issue stays the same. 
i Uncomment line 10 `import wordclock_interfaces.web_interface as wciweb` and line 94 `self.wciweb = wciweb.web_interface(self)
` 

now the wordclock.py works as expected. maybe i find a solution to use the interface in the future

FrankX0, 2020-09-28:
Would it help to have the clock run on Python 3?
I'm currently in the process of doing that.
Most of the plugins compile (not physically tested though), including the web interface.

nvanoost, 2020-10-11:
Anyone found a solution for this? Or how I can force install the werkzeug module in the right directory? I have exactly the same error on my Zero W

kas1kas, 2020-11-10:
I also have this problem but here the install path is different:
pi@raspberrypi:~/rpi_wordclock $ pip show werkzeug outputs
Name: Werkzeug
Version: 1.0.1
Summary: The comprehensive WSGI web application library.
Home-page: https://palletsprojects.com/p/werkzeug/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD-3-Clause
Location: /usr/local/lib/python2.7/dist-packages
Requires:

Did someone find a good fix already?

I first did the setup as suggested by nvanoost in the Updated package installation.

phenze, 2020-11-11:
I think you must run the pip commands also with sudo.
`sudo pip install 'Werkzeug==0.16.1'`

When this does not work try this:
https://stackoverflow.com/questions/28874685/pip-how-to-install-into-usr-local

kas1kas, 2020-11-11:
**Yes, that  worked but gave me another problem.
first:**
pi@raspberrypi:~ $ sudo pip install 'Werkzeug==0.16.1'
Collecting Werkzeug==0.16.1
  Downloading https://files.pythonhosted.org/packages/c2/e4/a859d2fe516f466642fa5c6054fd9646271f9da26b0cac0d2f37fc858c8f/Werkzeug-0.16.1-py2.py3-none-any.whl (327kB)
    100% |████████████████████████████████| 327kB 692kB/s
Installing collected packages: Werkzeug
  Found existing installation: Werkzeug 1.0.1
    Uninstalling Werkzeug-1.0.1:
      Successfully uninstalled Werkzeug-1.0.1
Successfully installed Werkzeug-0.16.1

**So, now the ImportError: cannot import name cached_property is gone but a new error appears:**

Update deprecated external dependency rpi_ws281x. For details see also https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md
Traceback (most recent call last):
  File "wordclock.py", line 166, in <module>
    word_clock = wordclock()
  File "wordclock.py", line 54, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 63, in __init__
    self.strip.begin()
AttributeError: wordclock_display instance has no attribute 'strip'

**Do I have to update the ws281? No idea how to do that. Any help for this new problem?**

FrankX0, 2020-11-11:
Please check issue #170 and see if this resolves your current issue.

kas1kas, 2020-11-11:
I finally got it running
Last error was caused by 

**git checkout 2f9e03c45ba0964029204db565ad9d6233b3a1a6**

in the install sequence:

cd ~
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
**git checkout 2f9e03c45ba0964029204db565ad9d6233b3a1a6**
sudo scons
cd ~/rpi_ws281x/python
sudo python setup.py install

If I leave it out all works (with werkzeug 0.16.1 as indicated before)

kas1kas, 2020-11-11:
to FrankX0 on #170 
I have no idea how to install/update with this info:
(but it wasn't neccesary anyway :-)

Looking at the code, this issue is caused by failing to import the Adafruit NeoPixel library.
This reported in your log output:
Update deprecated external dependency rpi_ws281x. For details see also https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md

Please follow these instructions: this should resolve your issue.

FrankX0, 2020-11-11:
Like described at the end of #170, the issue was caused by not having the library (correctly) installed.
The message generated by wordclock_display.py is misleading.
Glad it resolved your issue.

@MaxZimmermann99 @schweizschweiz @nvanoost: can this issue be closed?

bk1285, 2020-12-19:
Hi all,

thank you for reporting, discussing (+resolving?)!

Please reopen, if the issue should still not be resolved.

Bernd

closed 2020-12-19 by bk1285