bk1285/rpi_wordclock #161: colorful light up

nUmb3rfive, 2020-06-29:
Hi guys,
first of all, you did an awesome job creating this software! Thank you very much!
Unfortunately some issues are occurring in recent times. There appears sometimes a color pattern for 1 minute until the time changes (attached a picture). The LEDs also flicker in rainbow mode despite the use of a level shifter. Do you guys have any idea how to fix this issue?
[image]
Thanks for your help in advance. If you need more informations for troubleshooting do not hesitate to ask.
Cheers,
Nicola

FrankX0, 2020-07-02:
Always difficult to debug this from a distance.
All points to noise on the data line to the LED's, causing them to flicker or turn on, which is "corrected" when the actual time is written again. Good that you use a level shifter!

Did you try a different power-supply?
Can you make pictures (or drawing) of how your power (especially ground) and signal connections are running?

echo1483, 2020-07-03:
Hallo Nicola,
hat zwar nichts damit zu tun. Was hast du als Streuscheibe für die LED´s benutzt.
Danke, echo

nUmb3rfive, 2020-07-04:
No, it‘s actually difficult to find such a powerful power supply. I ordered this one directly in China. 
Specs of my power supply:
- Input Voltage: 100~240V AC
- Output Voltage: 5V DC
- Output Current: 10A
- Suggested Usage: LED display, communication

Thank you for your help! I really appreciate it!

@echo1483 : In my opinion PLEXIGLAS® of evonik works best as a diffusor for the wordclock. I used PLEXIGLAS® Resist WHITE 99532 GT, thickness: 0.5mm. I hope this helps.

More pictures:
[image]
[image]
[image][image]

echo1483, 2020-07-05:
Thank you for your prompt reply.
echo

phenze, 2020-07-06:
Take a look here:

* https://github.com/bk1285/rpi_wordclock/issues/110
* https://github.com/bk1285/rpi_wordclock/issues/106

Looks like you have no common ground.

bk1285, 2020-12-19:
Reopen, if not resolved.

closed 2020-12-19 by bk1285