bk1285/rpi_wordclock #162: Add custom location for sunrise/sunset

koko0815, 2020-07-06:
Hey folks,

any idea to use a custom location in the sunrise-plugin? I want to use my own one, because Berlin is to fare away, as we are located in the south of germany.

I find out that web-page https://astral.readthedocs.io/en/latest/
But I'm not able to integrat that in Bernds Code :-/

Any help will be appreciated.

TIA
Phil

BTW: We have some problems with the weather forecast, too. It seems that there is no connection to weather_dot_com or yahoo? Any ideas?

FrankX0, 2020-07-06:
Did you try to configure the location in the config file?
https://github.com/bk1285/rpi_wordclock/blob/1c3abac795e9b0698413b4f58fcb52512ba85b64/wordclock_config/wordclock_config.example.cfg#L75

koko0815, 2020-07-07:
Hello FrankX0,

yes I did. But there is a difference about nearly 30 minutes between Berlin and our location near Stuttgart...
That's the reason why I want to set a custom location.

http://pythonhosted.org//astral/#cities is not working any more. But I was looking for the astral cities in the link above.
There is no southern location which is next to us than berlin.

TIA
Phil

FrankX0, 2020-07-07:
Hi Phil,

Now I understand. The number of cities is indeed very limited.
In the Netherlands this is not such a big issue ;-) (I'm off by only 5 minutes)

The best way IMO would be to be able to specify not only a location but also coordinates in the config file.
Astral seems to support that. Let me see what I can do.

Regards,
Frank.

FrankX0, 2020-07-08:
Hi Phil,

Can you try following modifications in your clock?
So instead of being able to select a few cities, now you can set your own with coordinates and time zone.
So you need to modify your config file and two modifications to the sunrise plugin.
If you're happy with this I can make a pull request and make it backwards compatible.

Regards,
Frank.

https://github.com/bk1285/rpi_wordclock/blob/d523f817f03835c05782703795a66e5f0ebdaf8d/wordclock_config/wordclock_config.example.cfg#L76
Change to: `location = "Den Bosch", "Netherlands", 51.6978162,5.3036748, "Europe/Amsterdam"`

https://github.com/bk1285/rpi_wordclock/blob/d523f817f03835c05782703795a66e5f0ebdaf8d/wordclock_plugins/sunrise/plugin.py#L1
Change to: `import astral`

https://github.com/bk1285/rpi_wordclock/blob/d523f817f03835c05782703795a66e5f0ebdaf8d/wordclock_plugins/sunrise/plugin.py#L21
Change to: `self.astral_at_location = astral.Location(info = eval(config.get('plugin_' + self.name, 'location')))`

bk1285, 2020-12-19:
Hi @FrankX0

looks promising: Nice approach! I think I find the time within the next weeks to look into the various issues/PRs/etc. ...including this topic...

Best,
Bernd