bk1285/rpi_wordclock #163: Alternative for pywapi

plotaBot, 2020-07-09:
pywapi is not under active development (latest version from 2014) and both APIs are apparently discontinued.
I suggest using [openweathermap API](https://openweathermap.org/current). Their free plan with 60 calls/minute and a total of a million per month should be sufficient. Keep in mind that the weather.com API discontinued in January 2020 and nobody ever since created an Issue here...
Any other suggestions besides openweathermap?

FrankX0, 2020-07-10:
Apparently the weather forecast is not the main usage of the clock 😃 
I would be in favour of going to openweahermap. Seems to have all we need.
And then I also want to suggest to use this for sunrise/set, moon-phases, etc. So also replacing astral.
This resolves issue #162 and only requires one location in the config file.

Of course this requires quite some changes.
@bk1285 what's your opinion?

plotaBot, 2020-07-10:
I also had something similar in mind. 
How about we save the location to the env vars based on the public IP? Only if there is no user specified location (City/LatLon). Making this the default method at the very first execution of the wordclock is a nice improvement.

FrankX0, 2020-07-10:
Getting the location from the IP could be nice touch, but for me not so important.
Users need to add their API key for openweathermap to the config anyway. A small step is then to also specify the GPS location.

bk1285, 2020-12-19:
I'll take a closer look to openweahermap, but that all sounds very promising!

Best,
Bernd

FabianReister, 2020-12-21:
@bk1285 see https://github.com/FabianReister/rpi_wordclock/tree/feature/omw_weather_forecast  :)

This is based on @phenze 's master-python3 branch. Therefore, it won't be possible to directly open a PR for this repository (but you could cherry pick the changes).

bk1285, 2021-07-24:
@FabianReister Link is dead... https://github.com/FabianReister/rpi_wordclock/tree/feature/omw_weather_forecast

R0lleston, 2021-08-11:
Seems to be one of the last plugin issues to sort....would be great to have weather working

FrankX0, 2021-11-20:
Just added support for OpenWeatherMap to the develop branch.
Just in time for the merge?

commit e9c5284c (2021-11-30): OpenWeatherMap

Pywapi replaced by OpenWeatherMap.
Fixes #163
closed 2021-11-30 by bk1285