bk1285/rpi_wordclock #165: Can't install all dependencies

Guduche, 2020-07-28:
Hi,

I have a problem installing the dependencies. I can't install ttf-freefont, flask, flask_restplus, svgwrite and am2303_rpi.
Here is when I use the installation script :
```console
  Installing dependency ttf-freefont...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package ttf-freefont is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ttf-freefont' has no installation candidate
  Done.
  Installing dependency flask...
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package flask
  Done.
  Installing dependency flask_restplus...
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package flask_restplus
  Done.
  Installing python dependency svgwrite...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting svgwrite
  Downloading https://www.piwheels.org/simple/svgwrite/svgwrite-1.4-py2.py3-none-any.whl (66kB)
    100% |████████████████████████████████| 71kB 445kB/s
svgwrite requires Python '>=3.6' but the running Python is 2.7.16
  Done.
  Installing python dependency am2303_rpi...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting am2303_rpi
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/am2303-rpi/

  Done.
```

I was able to install flask and flask_restuplus using pip.

svgwrite wants me to use python3 so I don't know what to do because the project uses python2 right ?
And ttf-freefont and am2303_rpi are not found.

Can anyone please help me ?
I'm using a RPi Zero WH

Guduche, 2020-07-28:
svgwrite dropped python 2 support since version 1.4, so you need to install the version 1.3.1
```console
sudo pip install flask flask_restplus svgwrite==1.3.1
```
Also, as mentionned here, the package ttf-freefont is not available on buster, so you need the last raspbian version running on stretch available [here](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/)

I haven't found anything about am2303_rpi yet.

bk1285, 2020-12-19:
Reopen, if not resolved.

closed 2020-12-19 by bk1285