bk1285/rpi_wordclock #166: svgwrite requires Python '>=3.6' but the running Python is 2.7.16

Silence667, 2020-08-03:
Ich habe gestern und heute versucht die Software zum Laufen zu bekommen, aber ich stoße immer wieder auf die gleichen Probleme. Ich habe bezüglich Python die Version 3.6 installiert, aber weiterhin die Fehler. 
Komme leider gerade nicht weiter und poste hier meinen Verlauf. Ich hoffe ihr könnt mir helfen.
`root@raspberrypi:~# sudo pip install pytz astral feedparser pillow svgwrite freetype-py netifaces monotonic flask-restplus

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages (2020.1)
Collecting astral
  Using cached https://files.pythonhosted.org/packages/ff/5d/261f49822b7f77c265197d46cdbaaaaf06f458734e9ac55a140dd75bdf29/astral-1.10.1-py2.py3-none-any.whl
Collecting feedparser
  Using cached https://files.pythonhosted.org/packages/91/d8/7d37fec71ff7c9dbcdd80d2b48bcdd86d6af502156fc93846fb0102cb2c4/feedparser-5.2.1.tar.bz2
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (5.4.1)
Collecting svgwrite
  Using cached https://www.piwheels.org/simple/svgwrite/svgwrite-1.4-py2.py3-none-any.whl
svgwrite requires Python '>=3.6' but the running Python is 2.7.16
root@raspberrypi:~#
root@raspberrypi:~# cd ~
root@raspberrypi:~# git clone https://github.com/jgarff/rpi_ws281x.git
Cloning into 'rpi_ws281x'...
remote: Enumerating objects: 831, done.
remote: Total 831 (delta 0), reused 0 (delta 0), pack-reused 831
Receiving objects: 100% (831/831), 362.70 KiB | 653.00 KiB/s, done.
Resolving deltas: 100% (509/509), done.
root@raspberrypi:~# cd rpi_ws281x
root@raspberrypi:~/rpi_ws281x#
root@raspberrypi:~/rpi_ws281x# sudo scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Version version.h
CC      main.o
CC      mailbox.o
CC      ws2811.o
CC      pwm.o
CC      pcm.o
CC      dma.o
CC      rpihw.o
AR      libws2811.a
RANLIB  libws2811.a
LINK    test
scons: done building targets.
root@raspberrypi:~/rpi_ws281x# cd ~/rpi_ws281x/python
root@raspberrypi:~/rpi_ws281x/python# sudo python setup.py install
running install
running bdist_egg
running egg_info
creating rpi_ws281x.egg-info
writing rpi_ws281x.egg-info/PKG-INFO
writing top-level names to rpi_ws281x.egg-info/top_level.txt
writing dependency_links to rpi_ws281x.egg-info/dependency_links.txt
writing manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
reading manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
writing manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv6l-2.7
copying neopixel.py -> build/lib.linux-armv6l-2.7
running build_ext
building '_rpi_ws281x' extension
swigging rpi_ws281x.i to rpi_ws281x_wrap.c
swig -python -o rpi_ws281x_wrap.c rpi_ws281x.i
creating build/temp.linux-armv6l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-InigCj/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv6l-2.7/rpi_ws281x_wrap.o
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-InigCj/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-InigCj/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv6l-2.7/rpi_ws281x_wrap.o -L../. -lws2811 -lrt -o build/lib.linux-armv6l-2.7/_rpi_ws281x.so
creating build/bdist.linux-armv6l
creating build/bdist.linux-armv6l/egg
copying build/lib.linux-armv6l-2.7/_rpi_ws281x.so -> build/bdist.linux-armv6l/egg
copying build/lib.linux-armv6l-2.7/neopixel.py -> build/bdist.linux-armv6l/egg
byte-compiling build/bdist.linux-armv6l/egg/neopixel.py to neopixel.pyc
creating stub loader for _rpi_ws281x.so
byte-compiling build/bdist.linux-armv6l/egg/_rpi_ws281x.py to _rpi_ws281x.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying rpi_ws281x.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying rpi_ws281x.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying rpi_ws281x.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying rpi_ws281x.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
writing build/bdist.linux-armv6l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/rpi_ws281x-1.0.0-py2.7-linux-armv6l.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing rpi_ws281x-1.0.0-py2.7-linux-armv6l.egg
Copying rpi_ws281x-1.0.0-py2.7-linux-armv6l.egg to /usr/local/lib/python2.7/dist-packages
Adding rpi-ws281x 1.0.0 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv6l.egg
Processing dependencies for rpi-ws281x==1.0.0
Finished processing dependencies for rpi-ws281x==1.0.0
root@raspberrypi:~/rpi_ws281x/python# cd ~
root@raspberrypi:~# wget https://launchpad.net/python-weather-api/trunk/0.3.8/+download/pywapi-0.3.8.tar.gz
--2020-08-03 21:34:14--  https://launchpad.net/python-weather-api/trunk/0.3.8/+download/pywapi-0.3.8.tar.gz
Resolving launchpad.net (launchpad.net)... 91.189.89.223, 91.189.89.222, 2001:67c:1560:8003::8003, ...
Connecting to launchpad.net (launchpad.net)|91.189.89.223|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://launchpadlibrarian.net/166317636/pywapi-0.3.8.tar.gz [following]
--2020-08-03 21:34:15--  https://launchpadlibrarian.net/166317636/pywapi-0.3.8.tar.gz
Resolving launchpadlibrarian.net (launchpadlibrarian.net)... 91.189.89.229, 91.189.89.228, 2001:67c:1560:8003::8008, ...
Connecting to launchpadlibrarian.net (launchpadlibrarian.net)|91.189.89.229|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25166 (25K) [application/x-tar]
Saving to: ‘pywapi-0.3.8.tar.gz’

pywapi-0.3.8.tar.gz 100%[===================>]  24.58K  --.-KB/s    in 0.02s

2020-08-03 21:34:15 (1.44 MB/s) - ‘pywapi-0.3.8.tar.gz’ saved [25166/25166]

root@raspberrypi:~# tar -zxf pywapi-0.3.8.tar.gz
root@raspberrypi:~# rm pywapi-0.3.8.tar.gz
root@raspberrypi:~# cd pywapi-0.3.8
root@raspberrypi:~/pywapi-0.3.8# sudo python setup.py build
running build
running build_py
creating build
creating build/lib.linux-armv6l-2.7
copying pywapi.py -> build/lib.linux-armv6l-2.7
root@raspberrypi:~/pywapi-0.3.8# sudo python setup.py install
running install
running build
running build_py
running install_lib
copying build/lib.linux-armv6l-2.7/pywapi.py -> /usr/local/lib/python2.7/dist-packages
byte-compiling /usr/local/lib/python2.7/dist-packages/pywapi.py to pywapi.pyc
running install_egg_info
Writing /usr/local/lib/python2.7/dist-packages/pywapi-0.3.8.egg-info
root@raspberrypi:~/pywapi-0.3.8# cd ~
root@raspberrypi:~# git clone https://github.com/bk1285/rpi_wordclock.git
Cloning into 'rpi_wordclock'...
remote: Enumerating objects: 2855, done.
remote: Total 2855 (delta 0), reused 0 (delta 0), pack-reused 2855
Receiving objects: 100% (2855/2855), 25.82 MiB | 928.00 KiB/s, done.
Resolving deltas: 100% (1803/1803), done.
Checking out files: 100% (239/239), done.
root@raspberrypi:~# cd ~/rpi_wordclock
root@raspberrypi:~/rpi_wordclock# sudo python wordclock.py
Traceback (most recent call last):
  File "wordclock.py", line 3, in <module>
    import netifaces
ImportError: No module named netifaces
root@raspberrypi:~/rpi_wordclock#
`

phenze, 2020-08-04:
Hallo,
die Software läuft nur mit Python 2.7
Wie schon hier https://github.com/bk1285/rpi_wordclock/issues/165 und hier https://github.com/bk1285/rpi_wordclock/issues/160 erwähnt braucht es spezifische Versionen der einzelnen Packages.

Außerdem solltest du Raspbian stretch und nicht das aktuellste Buster verwenden.

Silence667, 2020-08-04:
Ich werde es mal probieren. Habe auch noch eine Übersicht der alten Versionen gefunden https://howchoo.com/g/nzc0yjzjy2u/raspbian-stretch-download 
Lade gerade die 2018-11-13-raspbian-stretch.zip runter und werde es damit mal testen.

phenze, 2020-08-04:
Da würde ich lieber die offizielle Quelle nehmen:

https://downloads.raspberrypi.org/raspbian/images/

bk1285, 2021-08-09:
Sollte mit dem port auf py3 erledigt sein.

closed 2021-08-09 by bk1285