bk1285/rpi_wordclock #167: flask-restplus requires pyrsistent, which requires Python '>=3.5'

mh0815, 2020-09-23:
Hi,
I have solved most dependency problems myself / with the help of the issues discussions here. But I don't know how to solve this one:
When I install flask-restplus, it has a dependency to pyrsistent (>=0.14.0). So it downloads latest version (0.17.2), which is not running on python 2.7. So I installed pyrsistent 0.16.0, which would satisfy the requirement and run on python 2.7. But when I install flask-restplus using
`sudo pip install  'flask-restplus==0.12.1' `
it ignores the latest version and still downloads the latest version - which fails due to python 2.7. 
Is there a way I can modify the flask-restplus requirements to pyrsistent>=0.14.0 <=0.16.0 ? 

The workaround from @schweizschweiz mentioned in issue #160 (uncommenting web api use in wordclock.py) works to start without errors, but actually I think the web interface is one of the coolest features in this wordclock - to get rid of the buttons.

FrankX0, 2020-09-28:
Would it help to have the clock run on Python 3?
I'm currently in the process of doing that.
Most of the plugins compile (not physically tested though), including the web interface.

Sofradex, 2020-09-29:
1. clean install stretch
2. first install pyrsistent (0.14.7)
3. than all of the rest. worked for me.

InnerChimp, 2021-03-08:
Unfortunately I seem to have the exact same problem. 
I did try the suggestion @Sofradex gave us with the clean install, but when trying to install flask-restplus, it always ends with the following:

`Collecting pyrsistent>=0.14.0 (from jsonschema->flask-restplus)
  Downloading https://files.pythonhosted.org/packages/7d/ae/90ddcf28fb8eee5d4990920586d2856342e42faa95f39223f0b9762ef264/pyrsistent-0.17.2.tar.gz (106kB)
    100% |████████████████████████████████| 112kB 2.0MB/s
pyrsistent requires Python '>=3.5' but the running Python is 2.7.13`

I tried a clean install of Stretch 2 times and installed Pyrsistent before te rest, but with nu luck. 

Any way to keep flask-restplus from trying to install a newer Pyrsistent version?

phenze, 2021-03-08:
Try updating to pyhton3 and use a actual raspberry os.

Further information how to do that you can find here:

https://github.com/bk1285/rpi_wordclock/pull/175

InnerChimp, 2021-03-08:
> 
> 
> Try updating to pyhton3 and use a actual raspberry os.
> 
> Further information how to do that you can find here:
> 
> #175

Thank you for your reply @phenze, but actually that's how I initially started. 
I installed the latest Raspberry OS and Python3, but I ran into issues when I followed the [software setup](https://rpi-wordclock.readthedocs.io/en/master/doc_software_setup.html) guide. If I remember correctly, I had issues with a lot of packages, including netifaces. 
After reading comments like "wordclock doesn't run on python 3, you need python 2.7" I re-installed everything using the latest Raspbian Stretch and Python 2.7.

But do I understand from your post @phenze that I can actually use Raspberry OS with Python 3? 
How can I install all the required packages for this version?

phenze, 2021-03-08:
Yes i am using latest raspberry os with python3.

Take a look here:
https://github.com/phenze/rpi_wordclock/tree/master-python3

There is a Python3 Section with install instructions in the readme file

InnerChimp, 2021-03-08:
Thank you very much. I will definitely try this!

HeerNMeester, 2021-03-08:
@InnerChimp , Definitely do that! I reinstalled my clock last wednesday and after 3 hours of trouble with the P2.7 version I switched to @phenze's version on P3 and got it running at the first try.

InnerChimp, 2021-07-18:
> 
> 
> Yes i am using latest raspberry os with python3.
> 
> Take a look here:
> https://github.com/phenze/rpi_wordclock/tree/master-python3
> 
> There is a Python3 Section with install instructions in the readme file

Sorry for not coming back earlier, but I guess life happened. Resulting in all the elements laying around, waiting for me to finish the project.
So I started earlier this weekend and hooked everything up, but unfortunately it doesn't seem to run. I followed the Python3 section and it seemed to install fine. 

When trying to run it, command line gives me the following:

`pi@raspberrypi:~/rpi_wordclock $ sudo python3 wordclock.py
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to english_specials.
Imported plugin 0: "tetris".
Imported plugin 1: "leds_off".
Imported plugin 2: "restart".
  INFO: No activate-flag set for plugin __pycache__ within the config-file. Will be imported.
No active exception to reraise
Failed to import plugin __pycache__!
Imported plugin 3: "sunrise".
Imported plugin 4: "test_strip".
Not found brigtness sensor value
Using brigtness sensor : False
  Selected "time_default" as default plugin
Imported plugin 5: "time_default".
Imported plugin 6: "rainbow".
Imported plugin 7: "time_matrix".
Imported plugin 8: "matrix".
Imported plugin 9: "shutdown".
Imported plugin 10: "time_as_words_dutch".
No module named 'pywapi'
Failed to import plugin weather_forecast!
Imported plugin 11: "ip_address".
Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
  INFO: No activate-flag set for plugin snake within the config-file. Will be imported.
Imported plugin 12: "snake".
  INFO: No activate-flag set for plugin time_in_seconds within the config-file. Will be imported.
Imported plugin 13: "time_in_seconds".
Imported plugin 14: "feed_parser".
Running plugin time_default.
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Unhandled exception in thread started by <bound method web_interface.threaded_app of <wordclock_interfaces.web_interface.web_interface object at 0xb5563590>>
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 61, in threaded_app
    self.app.run(host='0.0.0.0', port=port)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 990, in run
    run_simple(host, port, self, **options)
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 1052, in run_simple
    inner()
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 1005, in inner
    fd=fd,
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 848, in make_server
    host, port, app, request_handler, passthrough_errors, ssl_context, fd=fd
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/http/server.py", line 137, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
^CTraceback (most recent call last):
  File "wordclock.py", line 182, in <module>
    word_clock.run_forever()
  File "wordclock.py", line 172, in run_forever
    self.run()
  File "wordclock.py", line 144, in run
    self.runPlugin()
  File "wordclock.py", line 121, in runPlugin
    self.plugins[self.plugin_index].run(self.wcd, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_plugins/time_default/plugin.py", line 187, in run
    event = wci.waitForEvent(2)
  File "/home/pi/rpi_wordclock/wordclock_interfaces/event_handler.py", line 38, in waitForEvent
    self.__wait_for(lambda: self.event != self.EVENT_INVALID, seconds)
  File "/home/pi/rpi_wordclock/wordclock_interfaces/event_handler.py", line 81, in __wait_for
    self.condition.wait(waittime)
  File "/usr/lib/python3.7/threading.py", line 300, in wait
    gotit = waiter.acquire(True, timeout)
KeyboardInterrupt
pi@raspberrypi:~/rpi_wordclock $`

bk1285, 2021-08-09:
fixed on latest develop. Reopen, if issue persists.

closed 2021-08-09 by bk1285