bk1285/rpi_wordclock #169: Updated package installation

nvanoost, 2020-10-11:
Since this project is now a few years old and some modules are updated,
some commands from the manual are not correct anymore.
So I was thinking that it was a good idea to collect all the specific versions for all
the issues here and collect them, to make it easier to let other start with this project.
I'm using a Zero W, but I think those can be also used on other models.

Use 2019-04-08-raspbian-stretch-lite image (https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/)

```
sudo apt-get update
sudo apt-get install python-pip python-scipy scons git swig ttf-freefont
sudo pip install pyrsistent==0.14.7
sudo pip install 'flask-restplus==0.12.1' 
sudo pip install 'Flask==1.0.2'
sudo pip install pytz astral feedparser pillow freetype-py netifaces monotonic
sudo pip install svgwrite==1.3.1
```

OPTIONAL (not tested because i'm not using the temparature sensor)
`sudo pip install am2302_rpi`

```
cd ~
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
git checkout 2f9e03c45ba0964029204db565ad9d6233b3a1a6
sudo scons
cd ~/rpi_ws281x/python
sudo python setup.py install
```

(from here you can use the manual, starting from 'install pywapi' https://rpi-wordclock.readthedocs.io/en/master/doc_software_setup.html#rd-party-dependencies-git-repositories

InnerChimp, 2021-03-08:
Thank you for the information. However, I seem to get an error when trying to use the specified pyrsistent version number.

`      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1133, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: package directory 'find_namespace:' does not exist

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-1gSSEn/pyrsistent/`

Djangx, 2021-03-21:
Thanks updating the comments. But after trying I still have an issue that the 'scons' command can not be found. Has somebody any idea? :-)

phenze, 2021-03-22:
Take a look here. Try updating to python3.
There is no need to use outdated versions.
It is also not neccessary anymore to compile rpi_ws281x yourself. 

https://github.com/bk1285/rpi_wordclock/issues/167

R0lleston, 2021-07-20:
I'm finding the following with Phenzes python 3 version.
1) The seconds plugin doesnt work correct, the 10's works, the 1's displays  a changing line of pixels
2) The sleep function doesnt work (seems to be a throwback from the original code)
3) Astral feature doesnt work, shows the same incorrect sunset and sunrise value for any city
4) Weather / temp function isnt available.

While the clock works and installation is easier i think this version is still very much beta

bk1285, 2021-07-25:
@R0lleston Short update: Fixed 2 and 3 on latest develop. Working on 4.

Best,
Bernd

R0lleston, 2021-07-25:
@bk1285 is it safe to use your development version (utilizing the install instructions in 'development section' or should i hold off . Currently I'm using @phenze python3 version, but I want as much functionality as possible working. Or would you suggest staying with your master version ?

bk1285, 2021-08-09:
@R0lleston should be fine to use latest develop. This will be merged to master soon, finishing the python3 migration... :)

Best,
Bernd

bk1285, 2021-08-09:
Hi @nvanoost,

thanks for reaching out! I considered these dependencies when updating latest docs on develop as well as the new requirement-file which allows simple installation of all python packages with one single command.

* https://rpi-wordclock.readthedocs.io/en/develop/doc_software_setup.html
* https://github.com/bk1285/rpi_wordclock/blob/develop/requirements.txt

I'll therefore close this issue for now.

Best,
Bernd

closed 2021-08-09 by bk1285