bk1285/rpi_wordclock #170: wordclock_display instance has no attribute 'strip'

nvanoost, 2020-10-11:
I'm trying to run the wordclock for the first time but I get an error about an missing attribute 'strip'.
But where can I set that? The config example has no one and the error seems to be that the wordclock_display section needs to have one.

```
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
Update deprecated external dependency rpi_ws281x. For details see also https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md
Traceback (most recent call last):
  File "wordclock.py", line 166, in <module>
    word_clock = wordclock()
  File "wordclock.py", line 54, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 63, in __init__
    self.strip.begin()
AttributeError: wordclock_display instance has no attribute 'strip'
sys.excepthook is missing
lost sys.stderr
sys.excepthook is missing
lost sys.stderr
sys.excepthook is missing
lost sys.stderr
sys.excepthook is missing
lost sys.stderr
sys.excepthook is missing
lost sys.stderr
```

FrankX0, 2020-10-16:
Looking at the code, this issue is caused by failing to import the Adafruit NeoPixel library.
This reported in your log output:
_Update deprecated external dependency rpi_ws281x. For details see also_ https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md

Please follow these instructions: this should resolve your issue.

wiefeli, 2020-10-18:
I have the same issue. How do I resolve this?

nvanoost, 2020-10-18:
I'm trying to follow the steps which @FrankX0 has posted but I have still no luck.

FrankX0, 2020-10-19:
Can you try the following (should have already have been done by the install script):
```
cd /home/pi
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
sudo scons
cd /home/pi/rpi_ws281x/python
sudo python setup.py install
```

nvanoost, 2020-10-19:
o wow, tried that several times, but now it's working!
First I removed the whole rpi_ws281x directory with `sudo rm -rf ws_281x` and followd the steps again. Now the wordclock.py is working!

closed 2020-10-19 by nvanoost