bk1285/rpi_wordclock #172: wordclock does not run on startup via crontab

kas1kas, 2020-11-13:
wordclock does not start automatically after changing crontab:

**sudo crontab -e
@reboot sudo python /home/pi/rpi_wordclock/wordclock.py**

Any tip?
I have set console autologin as pi user
no message, no error

FrankX0, 2020-11-14:
This could be useful: [Debugging crontab jobs](https://stackoverflow.com/questions/4883069/debugging-crontab-jobs)

kas1kas, 2020-11-21:
I guess there is some bug in the wordclock software because it does work when I change

startup_message = ShowIP  to text.

Below the error with the ShowIP keyword:

I don't really need the IP address to be shown on startup but if anyone wants to fix it:

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 167, in <module>
    word_clock.startup()
  File "/home/pi/rpi_wordclock/wordclock.py", line 104, in startup
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
     * Environment: production
self.wcd.showText("IP: " + netifaces.ifaddresses(interface)[2][0]['addr'])

FrankX0, 2020-11-21:
Thanks for reporting this.
I expect that it fails because it does not have an ip address yet.
This could be handled in the application, or a delayed start could be introduced in crontab.
Or your solution of course.

bk1285, 2021-01-15:
Thanks for reporting, @kas1kas! The clock should definetly not crash, if no ip is provided.

bk1285, 2021-01-15:
Here we go, @kas1kas ;)

closed 2021-01-15 by bk1285