bk1285/rpi_wordclock #173: Word Clock difrent Layout

joeduck, 2020-11-17:
Hi,

first many thanks for that great work! Realy good Description and Documentation.
But still have a problem with the wiring. Cause of my Situation i have in front view the first LED (0) in the bottom right and the last LED of the grid in the upper left (109. Then follows the Minute LEDs as 110, 111, 112 and 113 in a center on buttom line.

In the wiring Config i have figured out the Minute LEDs all right. But the Words "ES IST" are starting in the Left bottom edge. Can you help me with the right formula for the coordinates?

Many thanks!

joeduck, 2020-11-18:
All LEDs working. But Words are wrong. Wiring like this scribble:
[image]

FrankX0, 2020-11-18:
Can you try the following (not tested):
```
        if x % 2 == 0:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT - y)
        else:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT + y - 1)
        return pos
```

joeduck, 2020-11-18:
Hi, thanks for your reply. This evening i read the Docs and the code and i begin to understand how to calculate the LED positions. The last few minutes i created a Excel sheet to calculate the formula.

Tried your formula, but it does sadly not work. I will do new tests.

Tanks again!

joeduck, 2020-11-18:
One Question: The LEDs on the Strip are countet form the begining of the Strip on the Pi with "0" to "109" on the grid, right?

FrankX0, 2020-11-18:
I think so: from 0 to (WIDTH x HEIGHT) - 1.
This "-1" is missing in my formula.
Can you try:
```
        if x % 2 == 0:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT - y) - 1
        else:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT + y - 1) - 1
        return pos

```

joeduck, 2020-11-18:
Thanks. But this does still not work. I must have in the upper left 2 LEDs "ES" then one off and then 3 LEDs "ist"...
Attached a Image.
[image]

joeduck, 2020-11-18:
your formula for the even columns is right, I calculated this with a few coloumns.
But the formula for the odd works not. we still try.

thanks a lot for your assistance.

joeduck, 2020-11-18:
with this formula it works in Excel. The Corordinates in Excel are right, but not on the Clock:
`
        if x % 2 == 0:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT - y) - 1
        else:
            pos = (self.WCA_WIDTH - x) * (self.WCA_HEIGHT + y ) - 10
        return pos
`
Calculate the Software the hole LEDs including the Minute LEDs for the Grid? Don´t understand :-(
Good night and many thanks for the help!

joeduck, 2020-11-19:
Nobody any hint? Must i doing an new wiring?

joeduck, 2020-11-19:
Have it :-)
`

        if x % 2 == 0:
            pos = (self.WCA_WIDTH - x) * self.WCA_HEIGHT - y - 1
        else:
            pos = (self.WCA_WIDTH - x) * self.WCA_HEIGHT + y - 10
        return pos
`

FrankX0, 2020-11-19:
Congrats!

joeduck, 2020-11-19:
Thanks FrankX0 for your time! :-)

FrankX0, 2020-11-20:
Now it is fixed: can you close this issue?
Thanx.

joeduck, 2020-11-20:
fixed :-)

closed 2020-11-20 by joeduck