bk1285/rpi_wordclock #174: Update code to Python 3x (Module wiring issue)

Ierlandfan, 2020-11-26:
I am trying to make the code work under Python3 so to make it more futureproof .

Things to change for Python3.

`sudo pip3 install Werkzeug==0.16.1` (Otherwise ImportError cached_Property)
`sudo pip3 install configparser `(no capitals) Otherwise ImportError ConfigParser)

Changes in wordclock.py
`import configparser `
(Instead of ConfigParser, otherwise Importerror configparser)
Line 36 
`self.config = configparser.configparser()`
(Instead of ConfigParser.ConfigParser()

`import _thread `( Instead of thread, otherwise ImportError thread)
 
Change web_interface.py (From line 139) to avoid syntax error

```
   if web_interface.app.wclk.developer_mode_active:
            channel_wise = lambda(x): {'red': x.r, 'green': x.g, 'blue': x.b}
        else:
            channel_wise = lambda(x): {'blue': x & 255, 'green': (x >> 8) & 255, 'red': (x >> 16) & 255}
```
 
Change to 

```
if web_interface.app.wclk.developer_mode_active:
            channel_wise = lambda x: {'red': x.r, 'green': x.g, 'blue': x.b}
        else:
            channel_wise = lambda x: {'blue': x & 255, 'green': (x >> 8) & 255, 'red': (x >> 16) & 255}
```
Probably forgot some minor issues but they can be sorted by using Google. 

The "only issue" I run into now is he "wiring" module.  (ImportError module wiring)

Does anybody knows the equivalent under python3?

plotaBot, 2020-11-26:
Have a look at #175 where i have some issues already tackeled. 
Tested this in developer mode and should run fine on a clock.

Ierlandfan, 2020-11-26:
Thanks for that! I works!  

Just a question: Do you need to install gtk3 and a window manager like Gnome to make it run in developer mode?

bk1285, 2021-01-15:
Hi @Ierlandfan 

Thanks for triggering the update to python 3. As you saw, @plotaBot started a migration already.... :)

For further discussion I'd therefore refer to #175 and close this issue.

developer mode is intended to run on a linux pc for development (this does actually not require a wordclock at all).

Best,
Bernd

closed 2021-01-15 by bk1285