bk1285/rpi_wordclock #176: Brightness and Sleeping not working

kas1kas, 2020-12-06:
The current build d523f81 still seems to suffer from two old problems:

1: 
Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section. 

2:
No sleeping time set, display will stay bright 24/7.
No sleeping time set, display will stay bright 24/7.
No sleep brightness set within the config-file. Defaulting to 5. 

Does someone have a solution? I tried all kind of suggestions but nothing works.

What I noticed:

Data for brightness in the config file is 200
Why does the code below jump to the except?  When I print the brightness value it says 200

        try:
            self.setBrightness(config.getint('wordclock_display', 'brightness'))
        except:
            self.brightness = max_brightness
            print(
                'Default brightness value not set in config-file: '
                'To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.')

Similar for the sleep times

thanks!!

Patrik42, 2021-01-02:
Hey! Same problem with my clock. changing the config file value of brightness does not change a thing.
Can we just change the wordclock.py so it does not need to draw the value out of the cfg file? 
I would just change the line to
self.setBrightness(config.getint('wordclock_display', 100)) 

would that work?

kas1kas, 2021-01-03:
Hard coding was also needed to get the sleeping time to work, but it does not change the fact that something is wrong in the total setup.
It all starts with the Raspberry OS install. 
I use raspbian stretch (instructions say to use the latest raspbian, but that does not work).
Then it needs corrections like manual versions:
pyrsistent==0.14.7
flask-restplus==0.12.1
flask==1.0.2
werkzeug==0.16.1

There are other forks claiming to use python 3, but sofar I have not found a clear step by step description that works!

Is there anyone who has everything working ? Please let me know how this was achieved.
Thanks a lot.

phenze, 2021-01-07:
Hi. I dont know if the brightness stuff is working in my fork because i have a brightness sensor installed.
But everything is working with python3 (There is a step by Step description in the Readme File)

Take a look here https://github.com/phenze/rpi_wordclock/tree/master-python3

kas1kas, 2021-01-09:
Hi Pascal,
Thanks  for your software: It installs very easy and without errors! Great.
The brightness and sleeping times however are still giving errors. But I will change the code as I did before.

Do you have the diagram for the light sensor that you use? 

**Thanks again!**

kas1kas, 2021-01-09:
Well, there is already a detailed description called Brightness sensor Issue #139 (sorry for not having seen that earlier).
I will try this.

closed 2021-01-09 by kas1kas

phenze, 2021-01-11:
Fell free to create a pull request when you have fixed that