bk1285/rpi_wordclock #178: Logging

FabianReister, 2020-12-21:
It is quite hard to identify errors on the command line as everything is printed without coloring. 

There are great libraries available, such as the **logging** library that allows to specify several logging levels. In conjunction with **coloredlogs**, the command line output would be

[image]

The only changes need to be made are

```
__main__:
  
  import logging
  logging.basicConfig(level=logging.DEBUG)
  
  import coloredlogs
  coloredlogs.install()
  print = logging.info
```
to wordclock.py' main function. This sets print to be an info message (available levels: DEBUG, INFO, WARNING, ERROR). 

**coloredlogs** can be installed from pypi (pip install coloredlogs) and could be wrapped with a try-catch-block to prevent import errors.

bk1285, 2021-01-15:
Nice, @FabianReister I'll consider that.

bk1285, 2021-01-15:
Thanks for the suggestion, @FabianReister :)

closed 2021-01-15 by bk1285