bk1285/rpi_wordclock #181: AttributeError: wordclock_display instance has no attribute 'strip'

teletext78, 2021-01-04:
Hello,

When I run "sudo python wordclock.py", I get this message:
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
Update deprecated external dependency rpi_ws281x. For details see also https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md
Traceback (most recent call last):
  File "rpi_wordclock/wordclock.py", line 166, in <module>
    word_clock = wordclock()
  File "rpi_wordclock/wordclock.py", line 54, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 63, in __init__
    self.strip.begin()
AttributeError: wordclock_display instance has no attribute 'strip'

What should I do?
Regards

FrankX0, 2021-01-05:
Did you check issue #170 ?
And especially https://github.com/bk1285/rpi_wordclock/issues/170#issuecomment-712172187

teletext78, 2021-01-05:
Thanks for your suggestion.
I did something completely different:
1) I follow this https://github.com/phenze/rpi_wordclock/tree/master-python3, since I also had issues with flask-restplus/Werkzeug;
2) the only error message when I run the program was then: "ModuleNotFoundError: No module named 'RPi'", which was easily solved with:
  sudo apt update
  sudo apt install python3-rpi.gpio

Now, everything seems to work!
I hope this could help somebody else as well.

Regards

closed 2021-01-05 by teletext78

phenze, 2021-01-07:
Thanks @teletext78 . I have added the missing package in my Readme