bk1285/rpi_wordclock #182: Wordclock doesn't light up right LED's

shruiik, 2021-01-05:
Hello there, 
I have setted up a Wordclock on my new Rpi.

since I have little space for the cables, I wrote a new wiring that looks like this: 

<table border=0 cellpadding=0 cellspacing=0 width=759 style='border-collapse: collapse;table-layout:fixed;width:575pt'>
 <tr>
  <td>Y/X</td>
  <td colspan=2>0</td>
  <td colspan=2>1</td>
  <td colspan=2>2</td>
  <td colspan=2>3</td>
  <td colspan=2>4</td>
  <td colspan=2>5</td>
  <td colspan=2>6</td>
  <td colspan=2>7</td>
  <td colspan=2>8</td>
  <td colspan=2>9</td>
  <td colspan=2>10</td>
 </tr>
 <tr>
  <td>0</td>
  <td>114</td>
  <td>E</td>
  <td>113</td>
  <td>S</td>
  <td>112</td>
  <td>K</td>
  <td>111</td>
  <td>I</td>
  <td>110</td>
  <td>S</td>
  <td>109</td>
  <td>T</td>
  <td>108</td>
  <td>L</td>
  <td>107</td>
  <td>F</td>
  <td>106</td>
  <td>Ü</td>
  <td>105</td>
  <td>N</td>
  <td>104</td>
  <td>F</td>

 </tr>
 <tr>
  <td>1</td>
  <td>93</td>
  <td>Z</td>
  <td>94</td>
  <td>E</td>
  <td>95</td>
  <td>H</td>
  <td>96</td>
  <td>N</td>
  <td>97</td>
  <td>Z</td>
  <td>98</td>
  <td>W</td>
  <td>99</td>
  <td>A</td>
  <td>100</td>
  <td>N</td>
  <td>101</td>
  <td>Z</td>
  <td>102</td>
  <td>I</td>
  <td>103</td>
  <td>G</td>

 </tr>
 <tr>
  <td>2</td>
  <td>92</td>
  <td>D</td>
  <td>91</td>
  <td>R</td>
  <td>90</td>
  <td>E</td>
  <td>89</td>
  <td>I</td>
  <td>88</td>
  <td>V</td>
  <td>87</td>
  <td>I</td>
  <td>86</td>
  <td>E</td>
  <td>85</td>
  <td>R</td>
  <td>84</td>
  <td>T</td>
  <td>83</td>
  <td>E</td>
  <td>82</td>
  <td>L</td>

 </tr>
 <tr>
  <td>3</td>
  <td>71</td>
  <td>T</td>
  <td>72</td>
  <td>G</td>
  <td>73</td>
  <td>N</td>
  <td>74</td>
  <td>A</td>
  <td>75</td>
  <td>C</td>
  <td>76</td>
  <td>H</td>
  <td>77</td>
  <td>V</td>
  <td>78</td>
  <td>O</td>
  <td>79</td>
  <td>R</td>
  <td>80</td>
  <td>J</td>
  <td>81</td>
  <td>M</td>

 </tr>
 <tr>
  <td>4</td>
  <td>70</td>
  <td>H</td>
  <td>69</td>
  <td>A</td>
  <td>68</td>
  <td>L</td>
  <td>67</td>
  <td>B</td>
  <td>66</td>
  <td>Q</td>
  <td>65</td>
  <td>Z</td>
  <td>64</td>
  <td>W</td>
  <td>63</td>
  <td>Ö</td>
  <td>62</td>
  <td>L</td>
  <td>61</td>
  <td>F</td>
  <td>60</td>
  <td>P</td>

 </tr>
 <tr>
  <td>5</td>
  <td>49</td>
  <td>Z</td>
  <td>50</td>
  <td>W</td>
  <td>51</td>
  <td>E</td>
  <td>52</td>
  <td>I</td>
  <td>53</td>
  <td>N</td>
  <td>54</td>
  <td>S</td>
  <td>55</td>
  <td>I</td>
  <td>56</td>
  <td>E</td>
  <td>57</td>
  <td>B</td>
  <td>58</td>
  <td>E</td>
  <td>59</td>
  <td>N</td>

 </tr>
 <tr>
  <td>6</td>
  <td>48</td>
  <td>K</td>
  <td>47</td>
  <td>D</td>
  <td>46</td>
  <td>R</td>
  <td>45</td>
  <td>E</td>
  <td>44</td>
  <td>I</td>
  <td>43</td>
  <td>R</td>
  <td>42</td>
  <td>H</td>
  <td>41</td>
  <td>F</td>
  <td>40</td>
  <td>Ü</td>
  <td>39</td>
  <td>N</td>
  <td>38</td>
  <td>F</td>

 </tr>
 <tr>
  <td>7</td>
  <td>27</td>
  <td>E</td>
  <td>28</td>
  <td>L</td>
  <td>29</td>
  <td>F</td>
  <td>30</td>
  <td>N</td>
  <td>31</td>
  <td>E</td>
  <td>32</td>
  <td>U</td>
  <td>33</td>
  <td>N</td>
  <td>34</td>
  <td>V</td>
  <td>35</td>
  <td>I</td>
  <td>36</td>
  <td>E</td>
  <td>37</td>
  <td>R</td>

 </tr>
 <tr>
  <td>8</td>
  <td>26</td>
  <td>W</td>
  <td>25</td>
  <td>A</td>
  <td>24</td>
  <td>C</td>
  <td>23</td>
  <td>H</td>
  <td>22</td>
  <td>T</td>
  <td>21</td>
  <td>Z</td>
  <td>20</td>
  <td>E</td>
  <td>19</td>
  <td>H</td>
  <td>18</td>
  <td>N</td>
  <td>17</td>
  <td>R</td>
  <td>16</td>
  <td>S</td>
 </tr>
 <tr>
  <td>9</td>
  <td>5</td>
  <td>B</td>
  <td>6</td>
  <td>S</td>
  <td>7</td>
  <td>E</td>
  <td>8</td>
  <td>C</td>
  <td>9</td>
  <td>H</td>
  <td>10</td>
  <td>S</td>
  <td>11</td>
  <td>F</td>
  <td>12</td>
  <td>M</td>
  <td>13</td>
  <td>U</td>
  <td>14</td>
  <td>H</td>
  <td>15</td>
  <td>R</td>
</table>

And my Wiring implementation looks like this:
```python
   class maiks_wiring:

    def __init__(self, WCA_WIDTH, WCA_HEIGHT):
        self.WCA_WIDTH = WCA_WIDTH
        self.WCA_HEIGHT = WCA_HEIGHT
        self.LED_COUNT = self.WCA_WIDTH * self.WCA_HEIGHT + 4

    def getStripIndexFrom2D(self, x, y):
        """
        Mapping coordinates to the wordclocks display
        Needs hardware/wiring dependent implementation
        Final range:
             (0,0): top-left
             (self.WCA_WIDTH-1, self.WCA_HEIGHT-1): bottom-right
        """
        if y % 2 == 0:
            pos = self.LED_COUNT - ((y * self.WCA_WIDTH) + (x))
        else:
            pos = self.LED_COUNT - (((y+1) * self.WCA_WIDTH) - (x+1))
        return pos

    def mapMinutes(self, min):
        """
        Access minutes (1,2,3,4)
        Needs hardware/wiring dependent implementation
        This implementation assumes the minutes to be wired as the last four leds of the led-strip
        """
        if min == 1:
            return 3 
        elif min == 2:
            return 2
        elif min == 3:
            return 1
        elif min == 4:
            return 0
        else:
            print('WARNING: Out of range, when mapping minutes...')
            print(min)
            return 0
```

Unfortunately, the wrong LEDs or no LEDs at all are lit.
Even if I choose another wiring, the correct number of LEDs is not lit.
I don't get any errors in the Console and when I debug, by using "print" lines, I get the solution that my mapping runs fine. But I don't get the right output. :(

I'm trying to solve this bug for 3 weeks now, but I can't find any solution. Can someone please help me.

phenze, 2021-01-07:
Did you have a common ground for your LED's and your RPI?
take a look here: https://github.com/bk1285/rpi_wordclock/issues/110

What is the output of strandtest example ?
https://github.com/bk1285/rpi_wordclock/issues/102
https://github.com/bk1285/rpi_wordclock/issues/106

With strandtest example file you can test which LED of your strip is causing an error.

bk1285, 2021-07-24:
@shruiik Please reopen, if not resolved.

closed 2021-07-24 by bk1285