bk1285/rpi_wordclock #183: Leds reset after each minute

wiefeli, 2021-02-27:
I made a wordclock and everything seems fine, but the Leds are blinking each minute. It seems like they are resetted each minute. Is there a way to enable the feature that only the "new" Leds light up and not a full reset every minute?

FrankX0, 2021-03-03:
What you are facing is not normal behavior. The LEDs are not supposed to blink every minute: the ones which should be off turn off, the ones which should be on turn on.
Maybe it’s related to your power-supply? Is the current rating sufficient?
Do you see the blinking only during the clock feature, or also with the other features (matrix, rainbow, etc.)?

wiefeli, 2021-03-03:
It's only with the time default plugin. The matrix, rainbow,.... are working perfect. It seems like checking the time every minute and therefore resetting each time

phenze, 2021-03-04:
Which Branch are you working on ?
Are you running on python2 or python3 ?
Which raspbian version are you using ?

I think your power supply is not enough at max power consumption:
- LED's : 114*0,06 = 6,84 A
- Raspberry 3: 1,8 A

On Top the PI 3 has a construction error which drops 0.6 V anywhere on the board. I don't know the exact problem.
But as far as i know the PI 3 is only working properly with the official power supply which has a power output of 5.2 V!
So try getting this official power supply and try powering the PI separately.

wiefeli, 2021-03-04:
But the LEDs need the max power only when all leds are up. 

However I think this is not the issue, since I had a similar setup with this Pi and it was working properly (but with older branch) but had to build everything from scratch again and now this issue appeared.

bk1285, 2021-03-04:
Tanks for reporting

Possibly caused here: https://github.com/bk1285/rpi_wordclock/blob/8e2e57fadb5b70c8e3de3a8df0c83fe144e30fbe/wordclock_plugins/time_default/plugin.py#L41

Requires a bug fix to allow further options than only fadeOutIn or typewriter... 

Regards,
Bernd

luchs7, 2021-04-20:
Thanks for the great work that has been done around the wordclock. 
I was also not happy with the fadeOutIn option and was happy to see the comment above.

The code to display the LEDs (wordclock_display) offers 3 options: typewriter, fadeoutin and no animation. 
The config file allows to set a varable typewriter to true/false (2 Options).
The plugin time_default allows only typewriter (true) and sets it to "fadeOutIn" otherwise.

As a workaround I changed the else value in the line mentioned above (anything other than typewriter or fadeOutIn) to trigger the 3rd option. E.g. self.animation = "typewriter" if self.typewriter else "None" 

Not a fix but the result is ok.