bk1285/rpi_wordclock #187: No module named flask_restplus

terryh01, 2021-05-15:
I am trying to run the wordclock script and I get the above error.
```
pi@raspberrypi:~/rpi_wordclock $ sudo python wordclock.py
Traceback (most recent call last):
  File "wordclock.py", line 11, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 4, in <module>
    from flask_restplus import Api, Resource, fields
ImportError: No module named flask_restplus
```

I have tried to run the install again for flask_restplus and this happens

```
pi@raspberrypi:~/rpi_wordclock $ sudo pip install flask-restplus
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting flask-restplus
  Using cached https://files.pythonhosted.org/packages/c2/a6/b17c848771f96ad039ad9e3ea275e842a16c39c4f3eb9f60ee330b20b6c2/flask_restplus-0.13.0-py2.py3-none-any.whl
Collecting jsonschema (from flask-restplus)
  Using cached https://files.pythonhosted.org/packages/c5/8f/51e89ce52a085483359217bc72cdbf6e75ee595d5b1d4b5ade40c7e018b8/jsonschema-3.2.0-py2.py3-none-any.whl
Requirement already satisfied: Flask>=0.8 in /usr/lib/python2.7/dist-packages (from flask-restplus) (1.0.2)
Requirement already satisfied: six>=1.3.0 in /usr/lib/python2.7/dist-packages (from flask-restplus) (1.12.0)
Collecting aniso8601>=0.82 (from flask-restplus)
  Using cached https://files.pythonhosted.org/packages/e3/04/e97c12dc034791d7b504860acfcdd2963fa21ae61eaca1c9d31245f812c3/aniso8601-9.0.1-py2.py3-none-any.whl
Requirement already satisfied: enum34; python_version < "3.4" in /usr/lib/python2.7/dist-packages (from flask-restplus) (1.1.6)
Collecting pytz (from flask-restplus)
  Using cached https://files.pythonhosted.org/packages/70/94/784178ca5dd892a98f113cdd923372024dc04b8d40abe77ca76b5fb90ca6/pytz-2021.1-py2.py3-none-any.whl
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from jsonschema->flask-restplus) (40.8.0)
Collecting attrs>=17.4.0 (from jsonschema->flask-restplus)
  Using cached https://files.pythonhosted.org/packages/20/a9/ba6f1cd1a1517ff022b35acd6a7e4246371dfab08b8e42b829b6d07913cc/attrs-21.2.0-py2.py3-none-any.whl
Collecting importlib-metadata; python_version < "3.8" (from jsonschema->flask-restplus)
  Using cached https://files.pythonhosted.org/packages/98/b8/8ec57a8ef46fbe7f185318c7ff7df9a06c9df451d9a59a067bfa851bb828/importlib_metadata-2.1.1-py2.py3-none-any.whl
Collecting pyrsistent>=0.14.0 (from jsonschema->flask-restplus)
  Using cached https://files.pythonhosted.org/packages/7d/ae/90ddcf28fb8eee5d4990920586d2856342e42faa95f39223f0b9762ef264/pyrsistent-0.17.2.tar.gz
pyrsistent requires Python '>=3.5' but the running Python is 2.7.16
```

Same problem when I try to run the wordclock script again. Can anyone help?

phenze, 2021-05-17:
There are several other issues here which already have a solution.

https://github.com/bk1285/rpi_wordclock/issues?q=is%3Aissue+flask+restplus

bk1285, 2021-08-09:
Please reopen, if not resolved.

closed 2021-08-09 by bk1285

R0lleston, 2021-08-09:
@bk1285 would it not be better to use Flaskrestx which seems to be updated more often and is working in @phenze Py3 variant of the clock ? From what I read Restplus is not being updated or is infrequently.

masete, 2023-03-06:
Flaskrestx is better because of the community around it.